firmadyne/firmadyne #87: No interfaces found for Asus firmware (system still boots)

smirani2, 2018-08-15:
Hello,

I have been testing FIRMADYNE with firmware for the Asus RT-AC3200, [version 3.0.0.4.382.50624](http://dlcdnet.asus.com/pub/ASUS/wireless/RT-AC3200/FW_RT_AC3200_300438250624.ZIP). The architecture is `armel`.

Everything works as expected until I run `inferNetwork.sh`, which is unable to find any interfaces.

A `run.sh` file is still generated, and I am able to login and run shell commands via the QEMU console; I just can't access the emulated system over the network. Here's the output of `inferNetwork.sh` (I increased the timeout):

```
$ ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
armel
Running firmware 1: terminating after 180 secs...
qemu-system-arm: terminating on signal 2 from pid 3503 (timeout)
Inferring network...
Interfaces: []
Done!
```

Once I start the system, I can see the following interfaces:

```
/ # ip a
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 52:54:00:12:34:59 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 52:54:00:12:34:58 brd ff:ff:ff:ff:ff:ff
4: eth2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 52:54:00:12:34:57 brd ff:ff:ff:ff:ff:ff
5: eth3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 52:54:00:12:34:56 brd ff:ff:ff:ff:ff:ff
6: tunl0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN 
    link/ipip 0.0.0.0 brd 0.0.0.0
7: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN 
    link/gre 0.0.0.0 brd 0.0.0.0
8: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
9: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN 
    link/sit 0.0.0.0 brd 0.0.0.0
10: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN 
    link/tunnel6 :: brd ::
```

Here is [qemu.initial.serial.log](https://github.com/firmadyne/firmadyne/files/2291549/qemu.initial.serial.log). There are a couple gpio-related errors, but nothing obviously fatal.

As I understand it, the infer script expects log entries matching "__inet_insert_ifa" in order to deduce the interface and IP address, but I don't see anything like that in the file. Is it possible the expected output is being suppressed by something?

Thanks for your time.

ddcc, 2018-08-15:
Unfortunately, the firmware doesn't seem to be configuring any of the network interfaces with an IP address, which is why firmadyne can't infer the network. You'll have to look through the startup process of the firmware to determine if some configuration variable is missing (perhaps there's some missing NVRAM key that specifies which is the default network interface), or if the startup process is aborting somewhere, etc.

DerZc, 2019-01-10:
@smirani2 Did you solve the problem? I meet the same problem and I do not know what to do