firmadyne/firmadyne #88: qemu-system-mips: Invalid parameter 'vlan'

joingh123, 2018-09-06:
Everything is OK until step.7
$ ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipseb
Running firmware 1: terminating after 60 secs...
qemu-system-mips: Invalid parameter 'vlan'

Help me please!

ddcc, 2018-09-06:
It looks like recent versions of QEMU have changed their networking syntax, which will need to be updated. For now, you could try an older version of QEMU.

joingh123, 2018-09-07:
> For now, you could try an older version of QEMU.

Thanks a lot for your reply !
So which version of QEMU shall I choose? 
How can I get an older version of QEMU?

Here is my apt-cache information about QEMU
```shell
$ apt-cache policy qemu-system-mips
qemu-system-mips:
  Installed: (none)
  Candidate: 1:2.5+dfsg-5ubuntu10.31
  Version table:
     1:2.5+dfsg-5ubuntu10.31 500
        500 http://us.archive.ubuntu.com/ubuntu xenial-updates/main i386 Packages
     1:2.5+dfsg-5ubuntu10.30 500
        500 http://security.ubuntu.com/ubuntu xenial-security/main i386 Packages
     1:2.5+dfsg-5ubuntu10 500
        500 http://us.archive.ubuntu.com/ubuntu xenial/main i386 Packages
```

ddcc, 2018-09-07:
I'm not sure where you obtained your current version of QEMU. The command line shows it is 2.12.50, and `apt-cache` shows that 2.5+dfsg-5ubuntu10.31 is available in the repository, but it is not installed. You might need to do some testing to see when the syntax was changed is; I'd guess somewhere around 2.11 or 2.12. Older versions are probably also available in the package repositories, at least for Debian/Ubuntu.

joingh123, 2018-09-07:
Thank you so much! @ddcc 
I tried qemu with v2.12.1:
```shell
$ git clone -b v2.12.1 https://github.com/qemu/qemu.git
```
It  works, though `-vlan` was deprecated in this version .

closed 2018-09-18 by joingh123

Zeus03, 2019-03-27:
> Thank you so much! @ddcc
> I tried qemu with v2.12.1:
> 
> ```shell
> $ git clone -b v2.12.1 https://github.com/qemu/qemu.git
> ```
> 
> It works, though `-vlan` was deprecated in this version .

Hey @ddcc 
I am trying to emulate firmware's using firmadyne. Got same error for ./scripts/inferNetwork.sh 4
I tried downgrading QEMU but its not happening. 
I tried following steps..

  -->  cloned the respective version (2.12 in my case)
  -->  mkdir build
  -- > cd build
  --> ../configure
  -->  make

Then i checked the version. < qemu-system-arm --version >
It shows..
QEMU emulator version 3.1.0 (Debian 1:3.1+dfsg-5)
Copyright (c) 2003-2018 Fabrice Bellard and the QEMU Project developers

Can you help me with this.

Thank you

ddcc, 2019-03-27:
It doesn't look like you've installed the version that you built (sudo `make install`)? Alternatively you could just use the pre-built packages for Debian stretch, which is still at 2.8: https://packages.debian.org/search?suite=stretch&searchon=names&keywords=qemu-system

Zeus03, 2019-03-28:
Thank you very much @ddcc .
I used the pre-built packages. First for qemu-system and then for different architectures.
Downgraded the version to 2.8...
Thank you again. :)

enamiirf, 2024-03-28:
$ sudo ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipseb
Running firmware 1: terminating after 60 secs...
qemu-system-mips: terminating on signal 2 from pid 14228 (timeout)
Inferring network...
Interfaces: []
Done!

enamiirf, 2024-03-28:
why there is no IP address