firmadyne/firmadyne #92: Repeating messages once started

jocko96b, 2018-10-04:
I managed to get my firmware running, and it somewhat appears to be running, but I am not sure if it is running correctly.  All get is the following repeated messages:

[  104.976000] firmadyne: ioctl: 0x94
[  104.976000] firmadyne: ioctl: 0x94
rtlioc: ret = -1
[  105.980000] firmadyne: ioctl: 0x94
[  105.980000] firmadyne: ioctl: 0x94
rtlioc: ret = -1
[  106.984000] firmadyne: ioctl: 0x94
[  106.984000] firmadyne: ioctl: 0x94
rtlioc: ret = -1
[  107.988000] firmadyne: ioctl: 0x94
[  107.988000] firmadyne: ioctl: 0x94
rtlioc: ret = -1
[  108.992000] firmadyne: ioctl: 0x94
[  108.992000] firmadyne: ioctl: 0x94
rtlioc: ret = -1
[  109.996000] firmadyne: ioctl: 0x94
[  109.996000] firmadyne: ioctl: 0x94
rtlioc: ret = -1

I am running a D-Link firmware: ftp://ftp2.dlink.com/PRODUCTS/DIR-645/REVA/DIR-645_FIRMWARE_1.01.ZIP

Any ideas?

ddcc, 2018-10-04:
If it runs fine (e.g. the web server is accessible, etc), you can ignore any error messages.

jocko96b, 2018-10-05:
That's the thing - I cannot get to the web server.  I do know my environment and network is functioning as I can run a different firmware and I can get to its web server just fine.  I'm just not sure if the firmware is janky (most likely) or if there is something that is not being handled properly (less likely)

ddcc, 2018-10-05:
It's most likely that this firmware isn't being fully emulated, because different firmware can have different behavior. The FAQ includes an entry about debugging that may be useful if you want to try to fix it.

jocko96b, 2018-10-15:
I found the issue.  You were correct that something wasn't being emulated properly.  I found that if I mount the image, /etc/init.d/S45gpiod.sh was trying to launch and it ended up bringing everything crashing down.  Disabling that script allowed everything to boot up for what I needed.

closed 2018-10-15 by jocko96b