firmadyne/firmadyne #93: Unable to access ports on emulated firmware from host OS

alomk, 2018-10-30:
These are my services running on the emulated firmware
[image]
I can ping from host and when I perform nmap on the guest ip I get this output 
[image]
I believe Firmadyne is redirecting guest ports to different ports on the host starting at 2001 but the labeled services are wrong. Running a more thorough scan for service fingerprinting returns nothing useful but I know one of the services is an http server. However, when I try connecting to any of the ports returned by nmap, my browser times out. I believe something is blocking the services on the guest from being accessed on the host, but I can still ping so I'm not sure. Do you have any thoughts on what the issue may be? Thanks

megele, 2018-10-30:
A firewall on the guest could easily have the effect you are describing. 

However, I'm pretty sure that Firmadyne does not "redirect" any ports. The emulated guest is simply attached to the tap network interface and all communications between the host and the guest are subject to the same rules and limitations that any two network-connected systems are.

What's strange in your screenshots is that the guest does not have the 200[1,2,3] ports open. Is it possible that you're nmap'ing a target that is something different than your guest?

cq674350529, 2018-10-31:
The ports 200[1,2,3] returned by `nmap` in host are not the one the emulated device listening. Instead, they seem to be the ports listening by your host machine ( related to the `Firmadyne`). 

[image]

In my case, I can connect to the port 80 from my host to the emulated device. 

[image]

[image]
 
But I come across another similar problem. When I copy `gdbsever` to the emulated device and attach to a process, I can't connect to the debug port in `gdb`, which the emulated decice is listening on. The result from nmap scan doesn't contain that port either.

I have tried many `gdbserver` of different versions, some of which "works" (means I can connect to the debug port from my host machine). But the gdb commands don't work (e.g. the attached process is still blocked after executing `continue` command in `gdb`).

ddcc, 2018-11-01:
I recall some network flakiness in the past, where sometimes communication works better in one direction than the other. You could play around with netcat and pipes to see if that works as a workaround. Another option would be to push the gdbserver connection through QEMU's serial connection, which eliminates the networking.

alomk, 2018-11-02:
You were right that I was nmapping the wrong target. The ip 192.168.1.2 is of my host machine through the tap interface. The ip I should be nmapping is the one returned by the infer network script right?
[image]
However, when I nmap this target (192.168.1.1) I receive no response. When I ping this ip from my host, I get no response as well. Something that is even more strange is that the ping response is coming from 192.168.1.2 which is my host. I have cleared all firewall rules on my guest and the issue persists. I've done testing with netcat and cannot communicate in either direction. I emulated the device from the repo's example and I have no problems with that so the issue is with the specific firmware I'm using (https://repo.turris.cz/archive/omnia/3.10.3/). Any thoughts as to what may be causing this?
[image]
[image]

HuangZhenyang, 2019-08-16:
There is some confusion about the result that why I can't get the `Network interfaces`

[image]

ddcc, 2019-08-16:
@HuangZhenyang Yours is different, please open a separate issue. I'm not familiar with Firmware Analysis Toolkit, you should try the instructions in the README of this repository. But it looks like the network isn't being inferred, which is why it doesn't show up online.

HuangZhenyang, 2019-08-16:
@ddcc  Thx! I'll see if there is some way to solve it. If I can't solve it, I will open a new issue.
But first, I think maybe it is because this Tenda firmware isn't suitable for this analysis toolkit. When I use the NETGEAR firmware, it works.

ddcc, 2019-08-16:
The emulation is very device specific. Depending on how it's implemented, some will emulate, some don't out of the box. The FAQ has some answers for debugging the firmware if you want to manually see if you can fix the networking.

HuangZhenyang, 2019-08-16:
I'll have a try, thx!   \(ﾟ▽ﾟ*)/

sssyyynnngithub, 2019-11-22:
ddcc Could you give me the detailed instruction on how to use gdbserver in firmadyne? 
 On the target qemu system, I use: ./gdbserver /dev/ttyS0 ./a.out , and on the other hand ,I use gdb-multiarch; set arch mips ; set endian little; set remotebaud 115200;target remote /dev/ttyS0. But it failed .Could you give me some detailed instructions ? Thank you !