firmadyne/firmadyne #94: Unable to execute the execute extractor.py

Minyanaing, 2018-11-09:
When I run the fat.py, the program start and then I put the firmware and the brand. Then the error occurred as described below. 
"""
Traceback (most recent call last):
  File "fat.py", line 122, in <module>
    main()
  File "fat.py", line 109, in main
    image_id = run_extractor(firm_name, firm_brand)
  File "fat.py", line 47, in run_extractor
    child = pexpect.spawn(extractor_cmd, timeout=None)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 204, in __init__
    self._spawn(command, args, preexec_fn, dimensions)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 276, in _spawn
    'executable: %s.' % self.command)
pexpect.exceptions.ExceptionPexpect: The command was not found or was not executable: /home/firmadyne/sources/extractor/extractor.py.
"""

cq674350529, 2018-11-09:
@Minyanaing Does the `extractor.py` file have executable permission?

ddcc, 2018-11-09:
I'm not familiar with `fat.py`. Have you tried following the instructions from the README?