firmadyne/firmadyne #100: Could not open mtd device: rootfs

E4ck, 2018-12-06:
When I was running ./scratch/1/run.sh, I got these errors:
init started:  BusyBox v1.4.2 (2018-04-25 18:28:18 CST) multi-call binary
[    5.988000] firmadyne: do_execve: /firmadyne/console
[    5.996000] OFFSETS: offset of pid: 0x100 offset of comm: 0x1f0
Unlocking rootfs ...
Could not open mtd device: rootfs
[    6.444000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 50006
[    6.512000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8200
mkdir: cannot create directory '/var/run': Input/output error
[    6.612000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8201
mkdir: cannot create directory '/var/log': Input/output error
[    6.688000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8202
mkdir: cannot create directory '/var/lock': Input/output error
[    6.780000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8203
mkdir: cannot create directory '/var/state': Input/output error
[    6.860000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8201
touch: /var/log/wtmp: Input/output error
[    6.940000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8201
touch: /var/log/lastlog: Input/output error
[    7.012000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 8206
touch: /tmp/resolv.conf.auto: Input/output error

I have check that the route was added successfully：
#route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.178.2   0.0.0.0         UG    20100  0        0 ens33
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 ens33
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 tap1_0
192.168.1.1     192.168.1.1     255.255.255.255 UGH   0      0        0 tap1_0
192.168.178.0   0.0.0.0         255.255.255.0   U     100    0        0 ens33

E4ck, 2018-12-06:
Forgot to say that I am running on ubuntu18

ddcc, 2018-12-06:
Does the emulation work otherwise? Certain errors are normal, though that one suggests that either some device-specific fixups are needed, or there may be some filesystem corruption. Does it make a difference if you restart from makeImage?

E4ck, 2018-12-07:
I think the simulation is not successful, I can't get the terminal, and I can't access the web page. I suspect that uhttpd didn't start successfully. I tried to use qemu simulation alone and found that the configuration file required by uhttpd is defective. May be this reason?

E4ck, 2018-12-07:
I think the simulation is not successful, I can't get the shell, and I can't access the web page. I suspect that uhttpd didn't start successfully. I tried to use qemu simulation alone and found that the configuration file required by uhttpd is defective. May be this reason?

ddcc, 2018-12-09:
Yeah, have you tried restarting from makeImage? Sometimes the QEMU disk image can get corrupted.