firmadyne/firmadyne #99: Most firmwares are able to be extracted?

hehe1397, 2018-12-06:


hehe1397, 2018-12-06:
I used the spiders to crawl a lot of firmware, but just a few firmware can be extracted. And fewer can be found the vulnerabilities. Is it normal?

cq674350529, 2018-12-06:
`binwalk` is great. But in general, there are many firmwares which `binwalk` can't handle. The most common situation is that when firmware is encrypted. In that situation, you have to decrypt the firmware first before analyzing it.

ddcc, 2018-12-06:
Yes, we had a fairly low success rate as well. I forget the exact numbers, but they are in the paper. Due to the diversity of devices, it's difficult to fully automate the entire process without more development work.

hehe1397, 2018-12-07:
> `binwalk` is great. But in general, there are many firmwares which `binwalk` can't handle. The most common situation is that when firmware is encrypted. In that situation, you have to decrypt the firmware first before analyzing it.

Thanks for your help, I'll try it

hehe1397, 2018-12-07:
> Yes, we had a fairly low success rate as well. I forget the exact numbers, but they are in the paper. Due to the diversity of devices, it's difficult to fully automate the entire process without more development work.

hehe1397, 2018-12-07:
> Yes, we had a fairly low success rate as well. I forget the exact numbers, but they are in the paper. Due to the diversity of devices, it's difficult to fully automate the entire process without more development work.

OK, I understand it. Thank you

closed 2018-12-07 by hehe1397