firmadyne/firmadyne #103: Receiving Error when trying to run tar2db

shilpkar-nitesh, 2019-03-09:
./scripts/tar2db.py -i 1 -f  "./images/1.tar.gz" 
Traceback (most recent call last):
  File "./scripts/tar2db.py", line 100, in <module>
    main()
  File "./scripts/tar2db.py", line 97, in main
    process(iid, infile)
  File "./scripts/tar2db.py", line 77, in process
    insertObjectToImage(iid, file2oid, links, cur)
  File "./scripts/tar2db.py", line 61, in insertObjectToImage
    for x in links])
psycopg2.IntegrityError: insert or update on table "object_to_image" violates foreign key constraint "object_to_image_oid_fkey"
DETAIL:  Key (oid)=(1) is not present in table "object".
-----------------------------------------------------------------------------------------------------------------

I checked the database the tables are present

firmware-> \dt
              List of relations
 Schema |      Name       | Type  |   Owner   
--------+-----------------+-------+-----------
 public | brand           | table | firmadyne
 public | image           | table | firmadyne
 public | object          | table | firmadyne
 public | object_to_image | table | firmadyne
 public | product         | table | firmadyne

cq674350529, 2019-03-11:
@shilpkar-nitesh  There is an foreign key constraint between table `object_to_image` and `object`. You should check the contents of table `object` and figure out why the `key(oid)=1` is not present in table `object`.

ddcc, 2019-03-12:
Have you tried clearing the database and restarting? I don't recall if the tar2db.py script actually needs to be run for regular emulation, and @cq674350529 is correct regarding the technical cause