firmadyne/firmadyne #105: extractor.py hangs on

dustinnoe, 2019-05-03:
```
/root/tmp/myFirmware.bin
>> MD5: 4e7933160ea07ebfa22b8d425afcffb1
>> Tag: 2
>> Temp: /tmp/tmpI0YwHL
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>>>> POSIX tar archive (GNU), owner user name: "_pkg_header.dat"
>> Recursing into archive ...
	
/tmp/tmpI0YwHL/_myFirmware.bin.extracted/uImage_add_header
	>> MD5: c219d5819a192b2dd6d10c98b8a3a722
	>> Tag: 2
	>> Temp: /tmp/tmpgzo_sT
	>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
	>> Recursing into archive ...
	>>>> Extraction failed!
	>>>> gzip compressed data, maximum compression, from Unix, last modified: 2015-03-04 08:38:15
	>> Recursing into compressed ...
		
/tmp/tmpgzo_sT/_uImage_add_header.extracted/40EC
		>> MD5: 8582b1f2e544fdd97c5bf5faeebb24e3
		>> Tag: 2
		>> Temp: /tmp/tmptly1YE
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>>>> ASCII cpio archive (SVR4 with no CRC), file name: "/dev", file name length: "0x00000005", file size: "0x00000000"
		>> Recursing into archive ...
```
I am attempting to extract a firmware but the process gets to this point and hangs with no errors. I have adjusted recursion depth and breadth with no effect. Any ideas?

ddcc, 2019-05-03:
Is the firmware available somewhere?

dustinnoe, 2019-05-04:
The firmware is available for download from the manufacturer. Thanks!
https://www.hanwha-security.com/en/products/video-recorder/nvr/ch4/SRN-472S/overview/

ddcc, 2019-05-06:
Since it's encrypted, the extractor script is going to fail, which is what happens. Perhaps you could trying updating binwalk?

```
./sources/extractor/extractor.py srn-472s-pkg_v1.06_150818163612.img -np -nk
```

```
/mnt/data/sources/firmadyne/srn-472s-pkg_v1.06_150818163612.img
>> MD5: 96c7cd5043bce2aca148d32fe6ce9833
>> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
>> Temp: /tmp/tmp4PcgT9
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>> Recursing into archive ...
>>>> Extraction failed!
>>>> gzip compressed data, from Unix, last modified: 2015-08-18 07:39:40
>> Recursing into compressed ...

/tmp/tmp4PcgT9/_srn-472s-pkg_v1.06_150818163612.img.extracted/0
        >> MD5: 62b10eeaf027780473b60f7b5b879e51
        >> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
        >> Temp: /tmp/tmpDvKlpg
        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
        >>>> POSIX tar archive, owner user name: "ders.23983/srn-472s_pkg_header.d", owner group name: "at"
        >> Recursing into archive ...

/tmp/tmpDvKlpg/_0.extracted/uImage_add_header
                >> MD5: 5e17acc7fff4f445557056e8e9790f90
                >> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
                >> Temp: /tmp/tmp1Hwygd
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >> Recursing into compressed ...
                >> Cleaning up /tmp/tmp1Hwygd...

/tmp/tmpDvKlpg/_0.extracted/logo_add_header.img
                >> MD5: 10be979132ec2cc7fd134ad032c5acbd
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >> Recursing into compressed ...
                >> Cleaning up /tmp/tmp1Hwygd...

/tmp/tmpDvKlpg/_0.extracted/logo_add_header.img
                >> MD5: 10be979132ec2cc7fd134ad032c5acbd
                >> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
                >> Temp: /tmp/tmprkcrz4
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >> Recursing into compressed ...
                >> Cleaning up /tmp/tmprkcrz4...

/tmp/tmpDvKlpg/_0.extracted/uRamdisk_add_header
                >> MD5: d0c5009634ff4f6589203a1855f59d49
                >> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
                >> Temp: /tmp/tmpDFyW6F
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >>>> gzip compressed data, ASCII, has header CRC, has 17241 bytes of extra data, last modified: 2014-10-29 08:52:13
                >> Recursing into compressed ...

/tmp/tmpDFyW6F/_uRamdisk_add_header.extracted/13C855E
                        >> MD5: d41d8cd98f00b204e9800998ecf8427e
                        >> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
                        >> Temp: /tmp/tmp7kcPGe
                        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                        >> Recursing into archive ...
                        >>>> Extraction failed!
                        >> Recursing into compressed ...
                        >> Cleaning up /tmp/tmp7kcPGe...
                >> Cleaning up /tmp/tmpDFyW6F...

/tmp/tmpDvKlpg/_0.extracted/u-boot_add_header.img
                >> MD5: 300b7d873937fc42ba41faf2c42bbfbc
                >> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
                >> Temp: /tmp/tmpDougl7
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >> Recursing into compressed ...
                >> Cleaning up /tmp/tmpDougl7...

/tmp/tmpDvKlpg/_0.extracted/srn-472s_pkg_header.dat
                >> MD5: 77226ac1a539b8fc7dd13f9e393c74e2
                >> Tag: srn-472s-pkg_v1.06_150818163612.img_96c7cd5043bce2aca148d32fe6ce9833
                >> Temp: /tmp/tmp2D7Dt4
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >> Recursing into compressed ...
                >> Cleaning up /tmp/tmp2D7Dt4...

/tmp/tmpDvKlpg/_0.extracted/uptools/pflash.ko
                >> MD5: 06cfa3de9fe5369778ba69b9d4acb873
                >> Skipping: application/x-object...
        >> Skipping: recursion breadth 5
        >> Skipping: completed!
        >> Cleaning up /tmp/tmpDvKlpg...

/tmp/tmp4PcgT9/_srn-472s-pkg_v1.06_150818163612.img.extracted/1743F2C
        >> MD5: d41d8cd98f00b204e9800998ecf8427e
        >> Skipping: d41d8cd98f00b204e9800998ecf8427e...
>> Cleaning up /tmp/tmp4PcgT9...

dustinnoe, 2019-05-06:
Indeed. I decrypted it before running it through the extractor. I cam PM the key and python if you are interested.

ddcc, 2019-05-07:
Can you just upload the decrypted firmware with simple reproduction steps somewhere? I can't take a look without a reproducer, and I don't have the time to decrypt the file or mess with it.

jlalvarezmedina, 2023-05-08:
> Indeed. I decrypted it before running it through the extractor. I can PM the key and python if you are interested.

Hey @dustinnoe, I would really appreciate if you can share with me the script for decrypting the SRN-472S firmware :)

efij, 2023-10-15:
Hey @dustinnoe, I would really appreciate if you can share with me the script/key for decrypting the SRN-472S firmware 
does it decrypt other hanwha products too?

VanIseghemThomas, 2024-01-29:
@ddc encountered the same problem using 2 seperate firmwares I was testing this tool against. Whenever it encounters an empty CPIO archive, it get's stuck at `>> Recursing into archive ...`.

I am able to manually extract the rootfs using binwalk in both cases, so it should be possible to extract those. I read somewhere it should be possible to manually use that one, but I'm not sure what to put into the database for this to work.

For context I'm running firmadyne using docker.
Example output:
```
root@0273aeb74a0d:~/firmadyne# ./sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk firms/GSM421xxx_MSM4214_XSM4216F_V13.0.4.19/M4250L-v13.0.4.19.stk images
>> Database Image ID: 3

/root/firmadyne/firms/GSM421xxx_MSM4214_XSM4216F_V13.0.4.19/M4250L-v13.0.4.19.stk
>> MD5: fe0c71f5ae9bee68db776b5cf5da37d5
>> Tag: 3
>> Temp: /tmp/tmpUNfc00
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>> Recursing into archive ...
>>>> Extraction failed!
>>>> gzip compressed data, maximum compression, from Unix, last modified: 1970-01-01 00:00:00 (null date)
>> Recursing into compressed ...
	
/tmp/tmpUNfc00/_M4250L-v13.0.4.19.stk.extracted/158
	>> MD5: c8c0717598619a9f9882db460f117bd6
	>> Tag: 3
	>> Temp: /tmp/tmp0lsi3l
	>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
	>>>> ASCII cpio archive (SVR4 with no CRC), file name: "dev", file name length: "0x00000004", file size: "0x00000000"
	>> Recursing into archive ...
```

ddcc, 2024-01-29:
It's been years since I looked at the code so you'll have to read the script for the details, but it should just put a <image id>.tar.gz file of the root filesystem into the images subdirectory.