firmadyne/firmadyne #106: soft lockup when enable do_execve hook

YourButterfly, 2019-05-16:
firmware DIR822A1_FW103WWb03  was emulated successfully,  but failed in  do_execve hook mode.
```txt
[    6.440000] firmadyne: do_execve[PID: 566 (rcS)]: argv: [ ! -f /etc/init0.d/S42pthrough.sh ], envp: HOME=/ TERM=vt102 LD_PRELOAD=/firmadyne/libnvram.so user_debug=31 PATH=/sbin:/usr/sbin:/bin:/usr/bin SHELL=/bin/sh USER=root
[    6.444000] firmadyne: do_execve[PID: 568 (svchlper)]: argv: sh /var/servd/PHYINF.ETH-2_start.sh, envp: SHELL=/bin/sh HOME=/ PATH=/sbin:/usr/sbin:/bin:/usr/bin
[    6.444000] firmadyne: do_execve[PID: 568 (svchlper)]: argv: sh /var/servd/PHYINF.ETH-2_start.sh, envp: SHELL=/bin/sh HOME=/ PATH=/sbin:/usr/sbin:/bin:/usr/bin
[    6.452000] firmadyne: do_execve[PID: 568 (svchlper)]: argv: sh /var/servd/PHYINF.ETH-2_start.sh, envp: SHELL=/bin/sh HOME=/ PATH=/sbin:/usr/sbin:/bin:/usr/bin
[    6.456000] firmadyne: do_execve[PID: 569 (S42pthrough.sh)]: argv: echo [/etc/init0.d/S42pthrough.sh]: start ..., envp: USER=root SHELL=/bin/sh PATH=/sbin:/usr/sbin:/bin:/usr/bin user_debug=31 LD_PRELOAD=/firmadyne/libnvram.so TERM=vt102 HOME=/
[    6.456000] firmadyne: do_execve[PID: 569 (S42pthrough.sh)]: argv: echo [/etc/init0.d/S42pthrough.sh]: start ..., envp: USER=root SHELL=/bin/sh PATH=/sbin:/usr/sbin:/bin:/usr/bin user_debug=31 LD_PRELOAD=/firmadyne/libnvram.so TERM=vt102 HOME=/
[    6.456000] firmadyne: do_execve[PID: 569 (S42pthrough.sh)]: argv:
[   19.636000] br0: port 1(eth0) entering forwarding state
[   84.828000] BUG: soft lockup - CPU#0 stuck for 61s! [sh:568]
[   84.828000] Modules linked in:
[   84.828000] Cpu 0
[   84.828000] $ 0   : 00000000 1000a400 0000023a 810bc700
[   84.828000] $ 4   : 8f07b710 8f07bb98 00000001 8f07b6e0
[   84.828000] $ 8   : 1000a401 1000001f 8f07bba0 ffffffff
[   84.828000] $12   : ffe2bdf9 8f07b710 00000001 8f07b6e0
[   84.828000] $16   : 00488a98 00000000 004731d8 00490000
[   84.828000] $20   : 00490000 00490000 00490000 00000007
[   84.828000] $24   : 8f07bb68 8067ab10                  
[   84.828000] $28   : 8f024000 8f025e60 7fe06f9b 8011eac0
[   84.828000] Hi    : 00000001
[   84.828000] Lo    : 810bc700
[   84.828000] epc   : 801011e0 handle_bp_int+0x18/0x58
[   84.828000]     Not tainted
[   84.828000] ra    : 8011eac0 kretprobe_trampoline+0x0/0xc
[   84.828000] Status: 1000a403    KERNEL EXL IE 
[   84.828000] Cause : 50808000
[   84.828000] PrId  : 00019300 (MIPS 24Kc)
```

ddcc, 2019-05-16:
Unfortunately, that's a known problem with the execve hook. As a workaround, you can [specify a value for the `firmadyne.syscall` that excludes execve](https://github.com/firmadyne/kernel-v4.1/blob/firmadyne-v4.1.17/drivers/firmadyne/hooks.c#L21).

YourButterfly, 2020-11-18:
As a temporary way to get the args of `exec*` ,  I modified function [do_execve](https://github.com/firmadyne/kernel-v2.6/blob/f32c9a3558c454b342de1c7b89a8fe7b2b10170b/fs/exec.c#L1379) last year.
If any progress, please tell me

closed 2020-11-18 by YourButterfly

ddcc, 2020-11-18:
The MIPS kernel was updated to a newer version in September, so you could test that and see if it's been fixed.