firmadyne/firmadyne #111: Question on scripts/tar2db.py

gal0is, 2019-06-16:
I have a question of the line 59 in the code. It seems insert oid as number 1 to all 'links' file. The ordinary file in 'files2oids' will use x[1] (correct oid in object table of the database) as oid in line 55 of the code. Why use the number 1 for all the 'link' file?

[https://github.com/firmadyne/firmadyne/blob/7de5bd8c8cbb9144919497dda67cdb353d1884c8/scripts/tar2db.py#L59](url)

```
55    cur.executemany(query, [{'iid': iid, 'oid' : x[1], 'filename' : x[0][0],
56                             'regular_file' : True, 'uid' : x[0][1],
57                             'gid' : x[0][2], 'mode' : x[0][3]} \
58                            for x in files2oids])
59    cur.executemany(query, [{'iid': iid, 'oid' : 1, 'filename' : x[0],
60                             'regular_file' : False, 'uid' : None,
61                            'gid' : None, 'mode' : None} \
62                            for x in links])
```

If there is no oid equal to 1 in the 'object' table of database(e.g. if you clear the 'object' table of the database rather than create a new one, the oid will not start from 1), it will cause error as follows.

```
Traceback (most recent call last):
  File "/root/deep/anaconda3/envs/iot/lib/python3.7/pdb.py", line 1701, in main
    pdb._runscript(mainpyfile)
  File "/root/deep/anaconda3/envs/iot/lib/python3.7/pdb.py", line 1570, in _runscript
    self.run(statement)
  File "/root/deep/anaconda3/envs/iot/lib/python3.7/bdb.py", line 585, in run
    exec(cmd, globals, locals)
  File "<string>", line 1, in <module>
  File "/root/deep/iot/firmadyne/firmadyne/scripts/tar2db.py", line 101, in <module>
    main()
  File "/root/deep/iot/firmadyne/firmadyne/scripts/tar2db.py", line 98, in main
    process(iid, infile)
  File "/root/deep/iot/firmadyne/firmadyne/scripts/tar2db.py", line 78, in process
    insertObjectToImage(iid, file2oid, links, cur)
  File "/root/deep/iot/firmadyne/firmadyne/scripts/tar2db.py", line 62, in insertObjectToImage
    for x in links])
psycopg2.IntegrityError: insert or update on table "object_to_image" violates foreign key constraint "object_to_image_oid_fkey"
DETAIL:  Key (oid)=(1) is not present in table "object".
```

Some guy have already encountered this error as #103.

megele, 2019-06-16:
Hi.

I have the following in the db for oid = 1

```
firmware=> select * from object where id = 1;
 id |          hash           |          created           |          lastupd           
----+-------------------------+----------------------------+----------------------------
  1 | dummy-hash-for-symlinks | 2015-07-29 21:15:30.348271 | 2015-07-29 21:15:30.348271
(1 row)
```
given the "hash" value this explanation suggests that the magic oid of 1 is used for symlinks.

gal0is, 2019-06-17:
> Hi.
>
> I have the following in the db for oid = 1
>
> ```
> firmware=> select * from object where id = 1;
>  id |          hash           |          created           |          lastupd [...]

Thanks for your reply. It's strange that your object table has 4 columns but only 2 in my table.
According to the database schema, the object table just has id and hash too.
[https://github.com/firmadyne/firmadyne#schema](url)

```
firmware=# select * from object where id = 1;
 id |               hash               
----+----------------------------------
  1 | 490d183785eebacb3653df4cde96544e
(1 row)
```
Maybe we are in different version and the old version has a preset row for oid = 1. 
Thanks.

megele, 2019-06-17:
Yes, you are right, that's very likely the case.

I would assume that in your/the official case, any symlinks reference a wrong object?

--m