firmadyne/firmadyne #112: ARM address space issue

canance, 2019-06-17:
Hi,

I have a custom ARM kernel that I created a profile for; however, I'm unable to get any plugins to work correctly.  

The kernel that I created the profile for was cross compiled using the musl compiler.  I cross compiled the kernel module in the same environment, used `dwarfdump` to create the `module.dwarf` file, and then created a zip containing `module.dwarf` and the `System.map` that was generated when I compiled the kernel.

To capture the memory, I used qemu's `dump-guest-memory` command.  Next, I tried to use the `linux_pslist` plugin to verify that my profile was working correctly.  I have attached the debug output from that command below.  

I see that it gets successfully identified as a `QemuCoreDumpElf` but then fails the `ArmAddressSpace` check.  

I am relatively new to volatility, so any advice or suggestions on how to fix this would be greatly appreciated!

Thanks,

Cory

### Debug output

```
root@0f7ae90f46d8:/app# vol.py --plugins=plugins --profile=Linuxarm_kernel-4_17ARM -f firmware_mem_dump1 --debug linux_pslist
Volatility Foundation Volatility Framework 2.6.1
DEBUG   : volatility.debug    : arm_kernel-4.17: Found dwarf file boot/System.map-4.1.17+ with 556 symbols
DEBUG   : volatility.debug    : arm_kernel-4.17: Found system file boot/System.map-4.1.17+ with 1 symbols
DEBUG   : volatility.debug    : Applying modification from BashHashTypes
DEBUG   : volatility.debug    : Applying modification from BashTypes
DEBUG   : volatility.debug    : Applying modification from BasicObjectClasses
DEBUG   : volatility.debug    : Applying modification from ELF32Modification
DEBUG   : volatility.debug    : Applying modification from ELF64Modification
DEBUG   : volatility.debug    : Applying modification from ELFModification
DEBUG   : volatility.debug    : Applying modification from HPAKVTypes
DEBUG   : volatility.debug    : Applying modification from LimeTypes
DEBUG   : volatility.debug    : Applying modification from LinuxIDTTypes
DEBUG   : volatility.debug    : Applying modification from LinuxTruecryptModification
DEBUG   : volatility.debug    : Applying modification from MachoModification
DEBUG   : volatility.debug    : Applying modification from MachoTypes
DEBUG   : volatility.debug    : Applying modification from MbrObjectTypes
DEBUG   : volatility.debug    : Applying modification from VMwareVTypesModification
DEBUG   : volatility.debug    : Applying modification from VirtualBoxModification
DEBUG   : volatility.debug    : Applying modification from LinuxGate64Overlay
DEBUG   : volatility.debug    : Applying modification from LinuxKmemCacheOverlay
DEBUG   : volatility.debug    : Requested symbol cache_chain not found in module kernel

DEBUG   : volatility.debug    : Applying modification from LinuxMountOverlay
DEBUG   : volatility.debug    : Applying modification from LinuxObjectClasses
DEBUG   : volatility.debug    : Applying modification from LinuxOverlay
DEBUG   : volatility.debug    : arm_kernel-4.17: Found dwarf file boot/System.map-4.1.17+ with 556 symbols
DEBUG   : volatility.debug    : arm_kernel-4.17: Found system file boot/System.map-4.1.17+ with 1 symbols
DEBUG   : volatility.debug    : Applying modification from BashHashTypes
DEBUG   : volatility.debug    : Applying modification from BashTypes
DEBUG   : volatility.debug    : Applying modification from BasicObjectClasses
DEBUG   : volatility.debug    : Applying modification from ELF32Modification
DEBUG   : volatility.debug    : Applying modification from ELF64Modification
DEBUG   : volatility.debug    : Applying modification from ELFModification
DEBUG   : volatility.debug    : Applying modification from HPAKVTypes
DEBUG   : volatility.debug    : Applying modification from LimeTypes
DEBUG   : volatility.debug    : Applying modification from LinuxIDTTypes
DEBUG   : volatility.debug    : Applying modification from LinuxTruecryptModification
DEBUG   : volatility.debug    : Applying modification from MachoModification
DEBUG   : volatility.debug    : Applying modification from MachoTypes
DEBUG   : volatility.debug    : Applying modification from MbrObjectTypes
DEBUG   : volatility.debug    : Applying modification from VMwareVTypesModification
DEBUG   : volatility.debug    : Applying modification from VirtualBoxModification
DEBUG   : volatility.debug    : Applying modification from LinuxGate64Overlay
DEBUG   : volatility.debug    : Applying modification from LinuxKmemCacheOverlay
DEBUG   : volatility.debug    : Requested symbol cache_chain not found in module kernel

DEBUG   : volatility.debug    : Applying modification from LinuxMountOverlay
DEBUG   : volatility.debug    : Applying modification from LinuxObjectClasses
DEBUG   : volatility.debug    : Applying modification from LinuxOverlay
Offset     Name                 Pid             PPid            Uid             Gid    DTB        Start Time
[... 81 lines cut ...]
 VMWareMetaAddressSpace: VMware metadata file is not available
 VirtualBoxCoreDumpElf64: ELF error: did not find any PT_NOTE segment with VBCORE
 VMWareAddressSpace: Invalid VMware signature: 0x464c457f
 MachOAddressSpace: MachO Header signature invalid
 LimeAddressSpace: Invalid Lime header signature
 WindowsHiberFileSpace32: PO_MEMORY_IMAGE is not available in profile
 WindowsCrashDumpSpace64BitMap: Header signature invalid
 WindowsCrashDumpSpace64: Header signature invalid
 HPAKAddressSpace: Invalid magic found
 VMWareMetaAddressSpace: VMware metadata file is not available
 VirtualBoxCoreDumpElf64: ELF Header signature invalid
 VMWareAddressSpace: Invalid VMware signature: -
 QemuCoreDumpElf: ELF Header signature invalid
 WindowsCrashDumpSpace32: Header signature invalid
 SkipDuplicatesAMD64PagedMemory: Incompatible profile Linuxarm_kernel-4_17ARM selected
 WindowsAMD64PagedMemory: Incompatible profile Linuxarm_kernel-4_17ARM selected
 LinuxAMD64PagedMemory: Incompatible profile Linuxarm_kernel-4_17ARM selected
 AMD64PagedMemory: Incompatible profile Linuxarm_kernel-4_17ARM selected
 IA32PagedMemoryPae: Failed valid Address Space check
 IA32PagedMemory: Failed valid Address Space check
 OSXPmemELF: ELF Header signature invalid
 FileAddressSpace: Must be first Address Space
 ArmAddressSpace: Failed valid Address Space check

```

ddcc, 2019-06-17:
Are you submitting this issue to the right repository? I've never used volatility, and I don't think we've ever tested it with firmadyne.

canance, 2019-06-17:
Hey @ddcc,

My apologies!  I did submit it to the wrong repo :-\.

Thanks,
Cory

closed 2019-06-17 by canance