firmadyne/firmadyne #113: ERROR: ld.so: object '/firmadyne/libnvram.so' from LD_PRELOAD cannot be preloaded (internal error): ignored.

yuroc5390, 2019-06-21:
Hi, I came across this issue when analyzing the firmware from Amcrest (things went well when I run firmadyne on other firmwares such as netgear). The libnvram.so couldn't be loaded. The firmware can be downloaded [here](https://s3.amazonaws.com/amcrest-files/Firmware/IP2M-841_IPM-721_V2.520.AC00.18.R.20170629.zip). I use binwalk to extract the root file system: rootfs.squashfs.img.

Here is what I did:
1.  
`./sources/extractor/extractor.py -b amcrest -sql 127.0.0.1 -np -nk rootfs.squashfs.img images`
>> Database Image ID: 1

/home/.../firmadyne/rootfs.squashfs.img
>> MD5: 88c6d45e196fce55e6d6990f24405815
>> Tag: 1
>> Temp: /tmp/tmpLuHfYr
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>> Recursing into archive ...
>>>> Squashfs filesystem, little endian, version 4.0, compression:xz, size: 1362284 bytes, 245 inodes, blocksize: 131072 bytes, created: 2017-06-29 16:17:33
>>>> Found Linux filesystem in /tmp/tmpLuHfYr/_rootfs.squashfs.img.extracted/squashfs-root!
>> Skipping: completed!
>> Cleaning up /tmp/tmpLuHfYr...
----------------------------------------------------------------------------

2.
`./scripts/getArch.sh ./images/1.tar.gz`
./bin/busybox: armel

3.
`./scripts/tar2db.py -i 1 -f ./images/1.tar.gz`
`sudo ./scripts/makeImage.sh 1`

Querying database for architecture... Password for user firmadyne: 
armel
----Running----
----Copying Filesystem Tarball----
----Creating QEMU Image----
Formatting '/home/...firmadyne/scratch//1//image.raw', fmt=raw size=1073741824
----Creating Partition Table----

Welcome to fdisk (util-linux 2.31.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xe7a413a5.

Command (m for help): Created a new DOS disklabel with disk identifier 0xf75b901b.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-2097151, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-2097151, default 2097151): 
Created a new partition 1 of type 'Linux' and of size 1023 MiB.

Command (m for help): The partition table has been altered.
Syncing disks.

----Mounting QEMU Image----
----Creating Filesystem----
mke2fs 1.44.1 (24-Mar-2018)
Discarding device blocks: done                            
Creating filesystem with 261888 4k blocks and 65536 inodes
Filesystem UUID: 794ec546-11fe-4363-94e7-22e00efc4750
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376

Allocating group tables: done                            
Writing inode tables: done                            
Writing superblocks and filesystem accounting information: done

----Making QEMU Image Mountpoint----
----Mounting QEMU Image Partition 1----
----Extracting Filesystem Tarball----
----Creating FIRMADYNE Directories----
[... 355 lines cut ...]

can't open /dev/myttyS0: No such file or directory

process '/etc/init.d/appauto' (pid 111) exited. Scheduling for restart.

can't open /dev/myttyS0: No such file or directory

process '/etc/init.d/appauto' (pid 112) exited. Scheduling for restart.

can't open /dev/myttyS0: No such file or directory
[   15.832155] cfg80211: Calling CRDA to update world regulatory domain

process '/etc/init.d/appauto' (pid 113) exited. Scheduling for restart.

can't open /dev/myttyS0: No such file or directory

process '/etc/init.d/appauto' (pid 114) exited. Scheduling for restart.

can't open /dev/myttyS0: No such file or directory

process '/etc/init.d/appauto' (pid 115) exited. Scheduling for restart.

can't open /dev/myttyS0: No such file or directory
[   18.982022] cfg80211: Calling CRDA to update world regulatory domain
QEMU: Terminated

ddcc, 2019-06-21:
Have you tried using the mount script to mount the firmware image, and verify that the libnvram.so is where it should be, and that it has a matching hash?

yuroc5390, 2019-06-21:
Thanks for the quick reply! I checked the image and firmadyne is at / .  And the files in /firmadyne are as follows:
`console`
`libnvram`
`libnvram.override`
`libnvram.so`
`preInit.sh`
`ttyS1`

I think the file is there. But I don't know how to verify the hash that you mentioned. Could you please instruct me how to do it? Thanks a lot!

ddcc, 2019-06-21:
Execute `md5sum <input path/libnvram.so>` and check that it matches `md5sum <firmadyne path/binaries/libnvram.so.armel>`

yuroc5390, 2019-06-23:
Thanks. The hash of libnvram.so.armel and "/firmadyne/libnvram.so" (in the qemu image) are the same.

ddcc, 2019-06-23:
Hmm, might be specific to that firmware then. This would probably require manual debugging to figure out the cause.

yuroc5390, 2019-06-24:
Thank you for your reply! Could you please share some approaches to debug this? I've tried using qemu -s -S ... and gdb-multiarch (by "target remote :1234"), but it seems that no debugging symbols can be found. And I don't know where to insert the breakpoint for the error.

ddcc, 2019-06-24:
Since the environment variable for LD_PRELOAD is supposed to be set by the kernel, you'll need to figure out why the dynamic linker (ld-linux) is emitting an error and not handling this correctly. Since there probably aren't debugging symbols included, you'll need to look at the disassembly, although having a decompiler like IDA will help. Alternatively, since the dynamic linker is probably open source, if you can figure out the approximate version, you could just rebuild it with debugging symbols using a cross-compile toolchain.

yuroc5390, 2019-06-24:
Cool, thanks! I'll give it a try.

yuroc5390, 2019-06-26:
Hi Dominic, the LD_PRELOAD issue was solved by compiling kernel and libnvram.so using the armhf toolchain. Because the ld.so in the firmware file system was compiled by armhf toolchain. Thanks for all the hints and help! I really appreciate it! Though there are other errors, I think this thread can be closed now.

ddcc, 2019-06-26:
Ah, that makes sense. Which armhf toolchain did you use?

yuroc5390, 2019-06-26:
It's  "arm-linux-gnueabihf-", directly installed by apt on Ubuntu 18.

closed 2019-06-27 by yuroc5390