firmadyne/firmadyne #114: pexpect.exceptions.TIMEOUT

LiRongbupt, 2019-07-17:
Traceback (most recent call last):
  File "./fat.py", line 122, in <module>
    main()
  File "./fat.py", line 114, in main
    arch = identify_arch(image_id)        
  File "./fat.py", line 62, in identify_arch
    child.expect("Password for user firmadyne: ")    
  File "/usr/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/usr/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python2.7/dist-packages/pexpect/expect.py", line 119, in expect_loop
    return self.timeout(e)
  File "/usr/lib/python2.7/dist-packages/pexpect/expect.py", line 82, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7f0af3bb95d0>
command: /root/firmadyne/scripts/getArch.sh
args: ['/root/firmadyne/scripts/getArch.sh', './images/1.tar.gz']
buffer (last 100 chars): '\xe7\x94\xa8\xe6\x88\xb7 firmadyne \xe7\x9a\x84\xe5\x8f\xa3\xe4\xbb\xa4\xef\xbc\x9a'
before (last 100 chars): '\xe7\x94\xa8\xe6\x88\xb7 firmadyne \xe7\x9a\x84\xe5\x8f\xa3\xe4\xbb\xa4\xef\xbc\x9a'
after: <class 'pexpect.exceptions.TIMEOUT'>

I follow up the setup instructions of Firmware Analysis Toolkit and I get this problem. I'm looking forward to your suggestions.Thank you very much

ddcc, 2019-07-17:
Sorry, I have no idea how Firmware Analysis Toolkit, you should report any problems with that to them. But, please try the firmadyne usage instructions and see if that works.

HuangZhenyang, 2019-08-15:
Same problem.