firmadyne/firmadyne #115: Nexthop has invalid gateway

aaronjeline, 2019-07-29:
Trying to test out this tool. The example firmware worked perfectly. Grabbed a D-link router firmware (DIR 645, 1.04.B11). When it try to emulate it under firmadyne, the run script gives the following: (I added set -e)
Adding route to 192.168.0.1...
+ sudo ip route add 192.168.0.1 via 192.168.0.1 dev tap1_0.1
Error: Nexthop has invalid gateway.
+ trap cleanup EXIT

And I can't reach the emulation over the network.

Any ideas?

cq674350529, 2019-07-30:
According to the commands you showed, it's invalid to specify `192.168.0.1` as the gateway, for the `192.168.0.1` is your target's addr. Maybe try with the addr `192.168.0.2`, which is the address of the `tap1_0.1`.

ddcc, 2019-07-31:
As @cq674350529 suggests, there may be a bug with the network inference, which you can work around by modifying the generated `run.sh` scrpt to call `ip route` with different arguments.

ddcc, 2019-07-31:
On second thought, this might be the VLAN bug mentioned in #77. Can you let me know if the solution in https://github.com/firmadyne/firmadyne/issues/77#issuecomment-384872909 works, and if so, I can commit the change.

muhammad-nouman-pakeventures, 2022-06-20:
Maybe the interface is down:

ip link set eno1 up
ip addr add 10.0.0.100 dev dno1
ip route add default via 10.0.0.1 dev dno1