firmadyne/firmadyne #116: Adding NVRAM values for new firmware?

paradoxymoron, 2019-08-01:
Hello!

I've been playing around with a Trust 18211 and after finding an S25FL032P on the board, I dumped its contents using an SPIDriver. Firmadyne was able to extract a whole bunch of neat stuff:

```
0             0x0             uImage header, header size: 64 bytes, header CRC: 0x95A81D66, created: 2011-05-31 08:33:07, image size: 107368 bytes, Data Address: 0x80200000, Entry Point: 0x80200000, data CRC: 0xA38FF64A, OS: Linux, CPU: MIPS, image type: Standalone Program, compression type: none, image name: "SPI Flash Image"
88288         0x158E0         U-Boot version string, "U-Boot 1.1.3 (May 31 2011 - 16:33:01)"
88912         0x15B50         CRC32 polynomial table, little endian
327680        0x50000         uImage header, header size: 64 bytes, header CRC: 0xD05836F9, created: 2012-03-23 08:20:49, image size: 3266809 bytes, Data Address: 0x80000000, Entry Point: 0x8023A000, data CRC: 0xEEC20F79, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "WA02"
327744        0x50040         LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 2448168 bytes
1112377       0x10F939        Squashfs filesystem, little endian, non-standard signature, version 3.0, size: 2480906 bytes, 517 inodes, blocksize: 65536 bytes, created: 2012-03-23 08:16:32
```

..and it managed to successfully construct a QEMU image.

However! `inferNetwork.sh` didn't find any network interfaces and when booting the image it eventually dies on a whole bunch of `nvram_get_buf: Unable to open key` warnings. Now I've read the FAQ and it suggested expanding `config.h` with the unknown values. So I went back to the dump, ran `strings` on it and found a whole bunch of key/value pairs that look like this:

```wan_ipaddr=192.168.1.1
wan_netmask=255.255.255.0
wan_gateway=192.168.1.254
wan_primary_dns=8.8.8.8
wan_secondary_dns=4.4.4.4
```

I _think_ I added them in the proper format to my own [config.h](https://github.com/firmadyne/firmadyne/files/3456952/config.h.txt) but booting the image again after recompiling `libnvram` doesn't result in any change.

Does anyone have a suggestion for fixing this?

paradoxymoron, 2019-08-01:
It has nothing to do with the issue but I figured this might give you a kick. :D

[image]

paradoxymoron, 2019-08-01:
Aah, I'm dumb! After casually browsing the filesystem I noticed the `libnvram.so` inside the image hadn't been updated, so I rebuilt the image and now my `config.h` changes have been applied.
Of course, now it's panicking, but that's a whole 'nother issue, at least I've made some progress. :D

closed 2019-08-01 by paradoxymoron

ddcc, 2019-08-01:
Glad you figured it out. Generally, I'd suggest using the filesystem method (https://github.com/firmadyne/libnvram) at `/firmadyne/libnvram.override`, since it's much easier to play around with NVRAM keys.