firmadyne/firmadyne #120: gdbserver is not working on mipseb kernel (WNAP320) Firmware -2

sssyyynnngithub, 2019-10-24:
I want to debug the firmware . But the gdbserver (7.12) can not start on the original kenel (2.6) . When I look other people's issues , I use the kenel (3.2) 32bit instead . But When I use this kernel , firmadyne cannot infer network : 
root@kali:~/firmadyne# ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne:
mipseb
Running firmware 1: terminating after 60 secs...
qemu-system-mips: -net nic,vlan=0: 'vlan' is deprecated. Please use 'netdev' instead.
qemu-system-mips: terminating on signal 2 from pid 1945 (timeout)
Inferring network...
Interfaces: []
Done!
and start the system failed : 
qemu-system-mips: -net nic,vlan=0: 'vlan' is deprecated. Please use 'netdev' instead.
[ 0.000000] Initializing cgroup subsys cpuset
[ 0.000000] Initializing cgroup subsys cpu
[ 0.000000] Linux version 3.2.0-4-4kc-malta (debian-kernel@lists.debian.org) (gcc version 4.6.3 (Debian 4.6.3-14) ) #1 Debian 3.2.51-1
[ 0.000000] bootconsole [early0] enabled
[ 0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[ 0.000000] FPU revision is: 00739300
[ 0.000000] Determined physical RAM map:
[ 0.000000] memory: 00001000 @ 00000000 (reserved)
[ 0.000000] memory: 000ef000 @ 00001000 (ROM data)
[ 0.000000] memory: 0067b000 @ 000f0000 (reserved)
[ 0.000000] memory: 0f894000 @ 0076b000 (usable)
[ 0.000000] debug: ignoring loglevel setting.
[ 0.000000] Wasting 60768 bytes for tracking 1899 unused pages
[ 0.000000] Initrd not found or empty - disabling initrd
[ 0.000000] Zone PFN ranges:
[ 0.000000] DMA 0x00000000 -> 0x00001000
[ 0.000000] Normal 0x00001000 -> 0x0000ffff
[ 0.000000] Movable zone start PFN for each node
[ 0.000000] early_node_map[1] active PFN ranges
[ 0.000000] 0: 0x00000000 -> 0x0000ffff
[ 0.000000] On node 0 totalpages: 65535
[ 0.000000] free_area_init_node: node 0, pgdat 806bfd60, node_mem_map 81000000
[ 0.000000] DMA zone: 32 pages used for memmap
[ 0.000000] DMA zone: 0 pages reserved
[ 0.000000] DMA zone: 4064 pages, LIFO batch:0
[ 0.000000] Normal zone: 480 pages used for memmap
[ 0.000000] Normal zone: 60959 pages, LIFO batch:15
[ 0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[ 0.000000] pcpu-alloc: [0] 0
[ 0.000000] Built 1 zonelists in Zone order, mobility grouping on. Total pages: 65023
[ 0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[ 0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[ 0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[ 0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[ 0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[ 0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[ 0.000000] Writing ErrCtl register=00000000
[ 0.000000] Readback ErrCtl register=00000000
[ 0.000000] Memory: 252252k/254544k available (4614k kernel code, 2292k reserved, 1283k data, 220k init, 0k highmem)
[ 0.000000] NR_IRQS:256
[ 0.000000] CPU frequency 200.00 MHz
[ 0.000000] Console: colour dummy device 80x25
[ 0.000000] Calibrating delay loop... 1087.48 BogoMIPS (lpj=2174976)
[ 0.020000] pid_max: default: 32768 minimum: 301
[ 0.020000] Security Framework initialized
[ 0.024000] AppArmor: AppArmor disabled by boot time parameter
[ 0.024000] Mount-cache hash table entries: 512
[ 0.028000] Initializing cgroup subsys cpuacct
[ 0.028000] Initializing cgroup subsys memory
[ 0.028000] Initializing cgroup subsys devices
[ 0.028000] Initializing cgroup subsys freezer
[ 0.028000] Initializing cgroup subsys net_cls
[ 0.028000] Initializing cgroup subsys blkio
[ 0.032000] Initializing cgroup subsys perf_event
[ 0.040000] devtmpfs: initialized
[ 0.048000] print_constraints: dummy:
[ 0.048000] NET: Registered protocol family 16
[ 0.056000] bio: create slab at 0
[... 298 lines cut ...]

Starting Translator... sh: cannot create /proc/sys/net/bridge/bridge-http-redirect-flush-mac: nonexistent directory
sh: cannot create /proc/sys/net/bridge/bridge-http-redirect-enabled: nonexistent directory
[http_redirect_tr]

Starting Translator... [dhcp]

Starting Translator... kill: cannot kill pid 611: No such process
[ntp]

Starting Translator... [timezone]

Starting Translator... kill: cannot kill pid 624: No such process
[sc_radio]
Error in opening the device.
: No such device

System initilization is .. [DONE...]

Welcome to SDK.

Have a lot of fun...

netgear123456 login:
Could anyone help me ?Thank you !

ddcc, 2019-10-25:
What's the problem with gdbserver? It's supposed to fairly portable when statically linked. Where'd you get kernel 3.2 from? We've only distributed 2.6.32 for MIPS.

sssyyynnngithub, 2019-11-22:
I get kernel 3.2 from https://people.debian.org/~aurel32/qemu/mips/. I see you said another way is to use serial port to debug . Could you give me some suggestion on how to gdb ? 
On the target qemu system, I use: ./gdbserver /dev/ttyS0 ./bin/ls, and on the other hand ,I use gdb-multiarch; set arch mips ; set endian little; set remotebaud 115200;target remote /dev/ttyS0. But it failed .Could you give me some detailed instructions ? Thank you !

ddcc, 2019-11-22:
The kernel needs to be modified for firmadyne, if you use a pre-built kernel without our patches, it will not emulate correctly. You can refer to the git commit log for our kernels to view the changes that we made. Sorry, we aren't actively maintaining firmadyne, and so I haven't tested the gdbserver integration in quite a few years. You will need to manually modify the QEMU command line to pass through another serial port into the guest, which you connect the gdbserver to.

sssyyynnngithub, 2019-11-22:
Sir, Did you ever tested  gdbserver on serial port successfully in kernel 2.6 with firmadyne?

ddcc, 2019-11-23:
No, I don't think I ever used the serial port to connect, it worked for me over the network. Working with the Linux kernel isn't straightforward, so if you haven't built a kernel from scratch before, or attempted to write a simple kernel module, I'd suggest that you start with something easier. Rebasing the kernel and updating the configuration would involve manually fixing any patches that fail to apply, such as those caused by changes to the internal kernel API.

amlamarra, 2020-04-27:
> You will need to manually modify the QEMU command line to pass through another serial port into the guest, which you connect the gdbserver to.

I thought this would be fairly straightforward, but I'm having some trouble. Could you provide an example `qemu-system-*` command to do this?

ddcc, 2020-04-27:
See e.g. #135

amlamarra, 2020-04-27:
> You can refer to the git commit log for our kernels to view the changes that we made.

Regarding this, it looks like your commits started on Feb 15, 2016. Is that correct? I'm going to use a newer kernel to get gdbserver working and I'd like to make sure I know I have all the commits you made.

ddcc, 2020-04-27:
Yeah, the commits are all rebased on top of the stable kernel

amlamarra, 2020-06-04:
I know it won't be fixed, but I also wanted to add that it seems `strace` also does not work in Firmadyne. I get the same results as `gdbserver`. I try running it (a statically linked binary) and nothing happens, forcing me to Ctrl-C. The same binary works fine on the physical device that I've emulated (linux 2.6.22). I was trying to use `strace` to see where `gdbserver` was getting hung up...

amlamarra, 2020-07-21:
So I'm still working on this issue. I tried cross-compiling `gdb` and running it on the emulated device. GDB works fine until you try to attach it to a PID. Considering this & the fact that `strace` doesn't work would imply that networking is not the issue. Maybe there's an issue with a certain syscall.

@ddcc Since you know a lot more about the Linux kernel than I do, would you have any idea on where to start looking?

ddcc, 2020-07-21:
I don't recall experiencing that issue at all, it seems pretty strange. You could look around the ptrace syscall options, which I believe are used by both gdb and strace. You could also build the kernel with kdb or add more prints to see what's going on when those programs hang.

amlamarra, 2020-07-24:
I don't think the issue is with `ptrace()`. I took a sample program from https://www.linuxjournal.com/article/6100, modified it to work on mips, cross-compiled + statically linked it, and it works on the Firmadyne-emulated device. Unless it's a particular "ptrace_request" that's not working. I also tried taking some ptrace source files from another mipsel device that uses the same kernel version (2.6.32) and replace Firmadyne's files.
These include:
- arch/mips/kernel/ptrace.c
- arch/mips/include/asm/ptrace.h
- include/linux/ptrace.h

I ran the device's image with this newly built kernel and got the same results. I also tried building the Firmadyne kernel w/out the Firmadyne module (same results).

I haven't tried building the kernel with KDB (have never used it before so there'll be a learning curve) or adding more prints. I wouldn't even know where to begin with the prints...

ddcc, 2020-07-25:
Hmm, that's strange. When you say that strace doesn't work, do you get no output at all? You could try enabling the debugging/verbose output flags of strace. I'm still suspecting something ptrace-related though. I believe both strace/gdb use ptrace_attach/ptrace_seize instead of ptrace_traceme, which the examples that you linked to seem to use. I don't think you should replace any of the firmadyne kernel source files, I don't recall modifying them so they should be identical. You could check the kernel configuration to see if there are any ptrace related options that are disabled, though.

For debugging, a more straightforward approach might be to enable QEMU's built-in gdbserver stub, and to connect a local gdb to that using the remote connection. Then, when the program hangs, you can hopefully get a full stack trace and look around to see where it's stuck.

amlamarra, 2020-07-28:
I've tried cross-compiling many different versions of `strace` with several different toolchains. Most of the older versions (<4.25) are unable to compile for various reasons. I've had the most success using the `gcc-8-mipsel-linux-gnu` Debian package. I'm doing this testing on 2 different mipsel devices. Testing multiple versions of strace, I've seen 2 different results. Sometimes, the same version gets me different results on the 2 devices.

Version 4.25, for example, will show this on one  device:
```
# strace -d ls
strace: ptrace_setoptions = 0x51
strace: PTRACE_SEIZE doesn't work
strace: new tcb for pid 191, active tcbs:1
strace: [wait(0x00177f) = 191] WIFSTOPPED,sig=SIGSTOP
strace: pid 191 has TCB_STARTUP, initializing it
strace: setting opts 0x51 on pid 191
strace: next_event: ignored SIGSTOP on pid 191

^Cstrace: cleanup: looking at pid 191
strace: Process 191 detached
strace: dropped tcb for pid 191, 0 remain
```
Ctrl-C will get me back to the prompt. And on the other device,  I'll see:
```
# strace -d ls
FATAL: kernel too old
[   49.680000] strace/746: potentially unexpected fatal signal 6.
[   49.680000] 
[   49.680000] Cpu 0
[   49.680000] $ 0   : 00000000 00000001 00000000 7fb0b764
[   49.680000] $ 4   : 00000003 7fb0b664 00000000 00000000
[   49.680000] $ 8   : 00000000 00000000 00000000 00000000
[   49.684000] $12   : 7fb0b6ec fffffffc 8ee3cc40 00000000
[   49.684000] $16   : 00000000 005bf318 7fb0b664 2aaa8000
[   49.684000] $20   : ffffffff 7fb0b8e0 00000005 005bf318
[   49.684000] $24   : 00000000 0046b180                  
[   49.684000] $28   : 005c8f80 7fb0b648 7fb0b8d8 004004d0
[   49.684000] Hi    : 00000005
[   49.684000] Lo    : 00000000
[   49.684000] epc   : 0046b260 0x46b260
[   49.688000]     Not tainted
[   49.688000] ra    : 004004d0 0x4004d0
[   49.688000] Status: 0000a413    USER EXL IE 
[   49.688000] Cause : 10800020
[   49.688000] PrId  : 00019300 (MIPS 24Kc)
Abort
```

From the little research I've done, if PTRACE_SEIZE doesn't work, it uses SIGSTOP. Judging by the output (`ignored SIGSTOP on pid 191`), that's not working either. Any idea why SIGSTOP wouldn't be working in the kernel?

Regarding QEMU's built-in gdbserver stub, won't that get me into the kernel's memory space? Should I check the kernel's stack trace with strace hung up?

ddcc, 2020-07-28:
It turns out that apparently PTRACE_SEIZE didn't land until 3.4, so that would explain why it doesn't work. From searching around, it seems that older kernel versions were known to have somewhat buggy ptrace implementations, e.g. https://stackoverflow.com/questions/39978621/ptrace-cant-work-well-in-kernel-2-6-for-mips-in-qemu . You could manually browse through the commit log to figure out what patches to backport to the kernel, but the easiest would probably be to rebase the firmadyne patches onto the latest 2.6.x kernel, e.g. 2.6.39.4.

amlamarra, 2020-07-28:
> the easiest would probably be to rebase the firmadyne patches onto the latest 2.6.x kernel, e.g. 2.6.39.4.

I tried that once with a different kernel version but ended up not being able to compile it. Not all of the changes were straightforward since the code was so different. So I would just like to verify before trying this. The Firmadyne kernel commit log looks like this:
[image]

Would I need to make all changes starting on Feb 15, 2016? Or would I start at Feb 17?

ddcc, 2020-07-28:
No, those are a bunch of cherry-picks from subsequent kernels that I added to implement various features or fix bugs. You'll want to rebase everything since the 2.6.32.70 release in the commit history onto the newer 2.6.39.4 branch. If any of those patches have already been applied, they'll automatically get dropped by git during the rebase. Something along the lines of `git rebase 2b00ed1 --onto <v2.6.39.4 branch name>`

amlamarra, 2020-07-29:
Ooookay, so I tried doing that manually last time... lol. I've never used `git rebase` so there was a bit of trial and error here. Here's what I eventually ended up doing:
```
git clone https://github.com/firmadyne/kernel-v2.6.32.git
git clone --branch v2.6.39.4 git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git
cd linux
git remote add firmadyne-v2.6.32.70 ~/kernel-v2.6.32
git fetch firmadyne-v2.6.32.70
git checkout firmadyne-v2.6.32.70
git rebase 2b00ed1 --onto v2.6.39.4
```
Then I had to manually fix a few few conflicts and `git add` the files. After fixing each conflict (I think there was 4), I could `git rebase --continue`.

However, this would not compile. There was an issue with the added `unionfs` support on the file `fs/unionfs/union.h`. I got a compiler error saying, `linux/smp_lock.h: No such file or directory`. According to [this StackExchange question](https://unix.stackexchange.com/questions/46540/fatal-error-linux-smp-lock-h-no-such-file-or-directory), that file does not exist as of v2.6.39. So I just copied that file from the Firmadyne kernel to this one & re-compiled. Then, it complains about a missing symbol definition, so I fix  that & re-compile. Another error: `/home/andrew/linux/fs/unionfs/file.c:310:9: error: too many arguments to function 'lower_inode->i_fop->fsync'`
That fsync function definition differs between the 2 kernel versions in the file `include/linux/fs.h`. So I change that and get another error: `/home/andrew/linux/ipc/shm.c:286:9: error: too few arguments to function 'sfd->file->f_op->fsync'`
Now it's messed up for other calls to `fsync()`. I'm not sure how deep this will go and how much time I'd spend on fixing errors only to create more errors.

Is unionfs support necessary? I was able to compile w/out it, but when trying to boot the device with the newly built kernel, it doesn't get very far:
```
Starting firmware emulation... use Ctrl-a + x to exit
[    0.000000] Linux version 2.6.39.4+ (andrew@kali) (gcc version 5.3.0 (GCC) ) #1 Tue Jul 28 19:36:17 EDT 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0063a000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f8d6000 @ 0072a000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 58688 bytes for tracking 1834 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x00010000
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x00010000
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 806c60b0, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60960 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0 init=/sbin/preinit
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252516k/254808k available (4336k kernel code, 2292k reserved, 1586k data, 236k init, 0k highmem)
[    0.000000] NR_IRQS:256
[    0.000000] CPU 0 Unable to handle kernel paging request at virtual address 00000080, epc == 806cc890, ra == 806cc870
[    0.000000] Oops[#1]:
[    0.000000] Cpu 0
[    0.000000] $ 0   : 00000000 10000000 1bdc0001 80700000
[    0.000000] $ 4   : 00000000 10000000 00000001 00000000
[    0.000000] $ 8   : fffffff8 8069f664 00000000 00000004
[    0.000000] $12   : 00000003 00000001 00000000 fffffffc
[    0.000000] $16   : 806a0000 80700000 806a0000 80700000
[    0.000000] $20   : 80700000 00000000 00000000 00000000
[    0.000000] $24   : 00000000 00000000                  
[    0.000000] $28   : 80696000 80697eb8 00000000 806cc870
[    0.000000] Hi    : 00000018
[    0.000000] Lo    : 00000000
[    0.000000] epc   : 806cc890 arch_init_irq+0x3c/0x14c
[    0.000000]     Not tainted
[    0.000000] ra    : 806cc870 arch_init_irq+0x1c/0x14c
[    0.000000] Status: 10000002    KERNEL EXL 
[    0.000000] Cause : 0080000c
[    0.000000] BadVA : 00000080
[    0.000000] PrId  : 00019300 (MIPS 24Kc)
[    0.000000] Modules linked in:
[    0.000000] Process swapper (pid: 0, threadinfo=80696000, task=8069ca00, tls=00000000)
[... cut ...]
[    0.000000]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    0.000000]         ...
[    0.000000] Call Trace:
[    0.000000] [<806cc890>] arch_init_irq+0x3c/0x14c
[    0.000000] [<806c97c8>] start_kernel+0x228/0x3d4
[    0.000000] [<80524cc0>] kernel_entry+0x0/0xa8
[    0.000000] 
[    0.000000] 
[    0.000000] Code: 8c445044  3c021bdc  24420001 <ac820080> 1000000b  24020001  8c645020  2402fffa  14820009 
[    0.000000] Disabling lock debugging due to kernel taint
[    0.000000] Kernel panic - not syncing: Attempted to kill the idle task!
```

ddcc, 2020-07-29:
If you don't plan to emulate any firmware using unionfs, then you can omit it.

That crash doesn't look right though, the kernel itself seems to be dereferencing a bad pointer to 0x80. No idea how that would be happening, you'd probably want to look at it in the debugger.

amlamarra, 2020-07-29:
I've started trying to figure that out. In the mean time, this looks like it may point me in the right direction:
https://lists.strace.io/pipermail/strace-devel/2014-January/002867.html

ddcc, 2020-07-29:
Hmm, that's strange. There's no difference at all with it applied?

amlamarra, 2020-07-29:
Correct.

I was able to confirm that this issue is not specific to the Firmadyne version of the Linux kernel. I cloned the Linux kernel tag v2.6.32.70, cross-compiled that to mipsel, and ran it with a standard debian image (https://people.debian.org/~aurel32/qemu/mipsel/). I have the same issue using strace on that (haven't tried gdb or gdbserver). Next, I'll try a slightly newer version, like v2.6.39.4 and do some more research to see which changes fixed this issue. But I probably won't be able to get to that until next week.

ddcc, 2020-07-30:
Thanks for follow up on this! Perhaps there's some other missing strace patches between 2.6.32.70 and 2.6.34?

With your previous attempt at 2.6.39.4 that wouldn't boot, was that with all the firmadyne patches and cherry-picks without unionfs?

amlamarra, 2020-07-30:
Correct.

And I spent a little time last night trying to build v2.6.39.4 for mipsel
using the musl toolchain you've provided and it just wouldn't work.

I did:
```
make mrproper
make defconfig
mkdir -p build/mipsel
cp .config build/mipsel/
make ARCH=mips
CROSS_COMPILE=/opt/cross/mipsel-linux-musl/bin/mipsel-linux-musl-
O=./build/mipsel/ -j $(nproc)
```

It didn't like something in the `arch/mips/Makefile` but I can't remember
what. I'm doing this from memory. I'll be looking into again on monday.

amlamarra, 2020-08-03:
Here's the full error when trying to cross compile with that musl toolchain:
```
...
  AS      arch/mips/kernel/r4k_fpu.o
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S: Assembler messages:
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:59: Error: opcode not supported on this processor: mips3 (mips3) `sdc1 $f0,272+0($4)'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:60: Error: opcode not supported on this processor: mips3 (mips3) `sdc1 $f2,272+16($4)'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:61: Error: opcode not supported on this processor: mips3 (mips3) `sdc1 $f4,272+32($4)'
...
make[3]: *** [/home/andrew/linux-2.6.39.4/scripts/Makefile.build:319: arch/mips/kernel/r4k_fpu.o] Error 1
make[2]: *** [/home/andrew/linux-2.6.39.4/scripts/Makefile.build:419: arch/mips/kernel] Error 2
make[1]: *** [/home/andrew/linux-2.6.39.4/Makefile:919: arch/mips] Error 2
make: *** [Makefile:130: sub-make] Error 2
```

There's a whole bunch of those `Error: opcode not supported on this processor` messages. I know the Firmadyne kernel had made some changes to that `r4k_fpu.S` file, so I copied that over, tried building again, but get slightly different errors:
```
...
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S: Assembler messages:
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:41: Error: unrecognized opcode `set_hardfloat'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:68: Error: unrecognized opcode `set_hardfloat'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:70: Error: unrecognized opcode `set_hardfloat'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:70: Error: opcode not supported on this processor: mips3 (mips3) `sdc1 $f0,272+0($4)'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:71: Error: unrecognized opcode `set_hardfloat'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:71: Error: opcode not supported on this processor: mips3 (mips3) `sdc1 $f2,272+16($4)'
/home/andrew/linux-2.6.39.4/arch/mips/kernel/r4k_fpu.S:72: Error: unrecognized opcode `set_hardfloat'
...
```

Any idea why that might be?

ddcc, 2020-08-03:
Looks like you will need to backport this toolchain fix for 2.6.39: https://www.linux-mips.org/archives/linux-mips/2015-07/msg00248.html

amlamarra, 2020-08-05:
I was able to get it to build by applying 3 of the Firmadyne MIPS commits to Linux v2.6.39.4:
[MIPS: Fix build with binutils 2.24.51+](https://github.com/firmadyne/kernel-v2.6.32/commit/3f738a85b285495376655789d96b024f92e297e6)
[MIPS: FPU: Fix conflict of register usage](https://github.com/firmadyne/kernel-v2.6.32/commit/74832107f8df7ab3da4f28b049c4fc9485c954dc)
[MIPS: Refactor 'clear_page' and 'copy_page' functions.](https://github.com/firmadyne/kernel-v2.6.32/commit/2b00f097459658150f46468e3dff5493f78a8670)

However, I get that same Kernel panic that I got earlier when I tried to rebase the Firmadyne commits onto v2.6.39.4.

ddcc, 2020-08-05:
Hmm, does it work if you skip the other firmadyne commits, so it's just those three on top of v2.6.39.4?

amlamarra, 2020-08-05:
Yup. Just those 3. Maybe there's something in the `.config` file. I think I was just using defaults.

ddcc, 2020-08-05:
Hmm, looks like an upstream kernel bug. You'll need to grab one more commit: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=639706

ddcc, 2020-08-05:
Can't promise you it works because I haven't built it, but since I've been doing some unrelated kernel work anyway, I've pushed a rebased branch for v2.6.39.4 without unionfs: https://github.com/firmadyne/kernel-v2.6.32/tree/firmadyne-v2.6.39.4

amlamarra, 2020-08-07:
Wow, it works!

So `strace` does work for that kernel, but only version 4.5.20. Other versions I've tried (both older & newer) get me that `FATAL: kernel too old` error. And one of the devices that I'm emulating will give me that error for every version of strace.

And `gdbserver` version 7.12 also works! I tried newer versions but would get `A problem internal to GDB has been detected` when trying to attach to it from GDB. I've only tried this on one device so far, but this is great!

ddcc, 2020-08-07:
Great, let me know if you run into any issues or regressions. If it's fine after some testing, I can upload binaries as a beta release.

amlamarra, 2020-08-08:
Cool, I'll keep testing it. What, exactly, did you do? Just rebase it without unionfs support?

ddcc, 2020-08-08:
Yeah, just rebase onto 2.6.39.4, skip unionfs, cherry-pick the malta fix plus another MIPS fix that came up when searching for that paging fault, and then reorganize everything chronologically.

agadient, 2020-09-01:
I'm running into the gdb problem as well. @amlamarra is the testing completed?

amlamarra, 2020-09-01:
It's been working for me. Just use the [firmadyne-v2.6.39.4](https://github.com/firmadyne/kernel-v2.6.32/tree/firmadyne-v2.6.39.4) branch of the kernel-v2.6.32 repo.

ddcc, 2020-09-01:
I uploaded some binaries to https://github.com/firmadyne/kernel-v2.6.32/releases/tag/v1.1 . Can you check if those work fine? If so, I'll update the repository to default to those

amlamarra, 2020-09-03:
Just finished testing them, both mipsel and mipseb work fine with certain versions of strace & gdbserver.

ddcc, 2020-09-03:
Ok, I will close this and consider it fixed by d1d1ffb94ead98eb471dfdf4511b514c08a3099a . Thanks!

closed 2020-09-03 by ddcc