firmadyne/firmadyne #121: Sir,Could you show the exact firmware version that firmadyne can simulate?

sssyyynnngithub, 2019-11-20:
Sir, I see you mentioned that "We evaluate FIRMADYNE on a real-world dataset of 23,035
firmware images across 42 device vendors gathered by our system.
Using a sample of 74 exploits on the 9,486 firmware images that
our system can successfully extract, we discover that 887 firmware
images spanning at least 89 distinct products are vulnerable to one
or more of the sampled exploit(s). "
Could you show the exact firmware version that firmadyne can simulate?
I use firmadyne but I can only simulate the only Netgear WNAP320! 
I simulate DIR-850L_REVB_MIDDLE_FIRMWARE_ONLY_v2.10B03,DIR815B1_FW203b08,DIR605,DIR890,DIR880,but all of them failed ! 
Could you give me some suggestion ? Thank you !
Thank you very much !

sssyyynnngithub, 2019-11-21:
here is the result about WNAP210_V2.1.4.tar:
[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:44:57 UTC 2016

[    0.000000] 

[    0.000000] LINUX started...

[    0.000000] bootconsole [early0] enabled

[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)

[    0.000000] FPU revision is: 00739300

[    0.000000] Determined physical RAM map:

[    0.000000]  memory: 00001000 @ 00000000 (reserved)

[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)

[    0.000000]  memory: 00606000 @ 000f0000 (reserved)

[    0.000000]  memory: 0f90a000 @ 006f6000 (usable)

[    0.000000] debug: ignoring loglevel setting.

[    0.000000] Wasting 57024 bytes for tracking 1782 unused pages

[    0.000000] Initrd not found or empty - disabling initrd

[    0.000000] Zone PFN ranges:

[    0.000000]   DMA      0x00000000 -> 0x00001000

[    0.000000]   Normal   0x00001000 -> 0x00010000

[    0.000000] Movable zone start PFN for each node

[    0.000000] early_node_map[1] active PFN ranges

[    0.000000]     0: 0x00000000 -> 0x00010000

[    0.000000] On node 0 totalpages: 65536

[    0.000000] free_area_init_node: node 0, pgdat 806923c0, node_mem_map 81000000

[    0.000000]   DMA zone: 32 pages used for memmap

[    0.000000]   DMA zone: 0 pages reserved

[    0.000000]   DMA zone: 4064 pages, LIFO batch:0

[    0.000000]   Normal zone: 480 pages used for memmap

[    0.000000]   Normal zone: 60960 pages, LIFO batch:15

[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024

[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0

[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)

[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)

[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)

[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.

[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes

[    0.000000] Writing ErrCtl register=00000000
[... 1621 lines cut ...]
[  100.744000] firmadyne: ioctl: 0x95

[  101.244000] firmadyne: ioctl: 0x94

[  101.296000] firmadyne: ioctl: 0x94

[  101.752000] firmadyne: ioctl: 0x95

[  102.248000] firmadyne: ioctl: 0x94

[  102.300000] firmadyne: ioctl: 0x94

[  102.760000] firmadyne: ioctl: 0x95

[  103.252000] firmadyne: ioctl: 0x94

[  103.304000] firmadyne: ioctl: 0x94

[  103.768000] firmadyne: ioctl: 0x95

[  104.280000] firmadyne: ioctl: 0x94

[  104.312000] firmadyne: ioctl: 0x94

\00[  104.780000] firmadyne: ioctl: 0x95

sssyyynnngithub, 2019-11-21:
I think the main problem is here:
DEVCONF: unable to read config data (ret=-3) 
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!     Uable to read device config.     !!
!! Setting is reset to factory default. !!

ddcc, 2019-11-22:
Sorry, we no longer have the database available. The example firmware listed in the usage instructions should work, but since we are not actively maintaining firmadyne anymore, we haven't tested it in quite some time. As for resolving issues with emulation, manual reversing may be needed, which we discuss in the FAQ and issue tracker.