firmadyne/firmadyne #123: Some routers execute ./scripts/inferNetwork.sh cannot get the IP address

0xbird, 2019-12-09:
I tested a lot of routers, why ca n’t I get the IP address?

birdpwn@ubuntu:~/fuzzer/firmadyne$ ./scripts/inferNetwork.sh 6
Querying database for architecture... Password for user firmadyne: 
mipseb
Running firmware 6: terminating after 60 secs...
qemu-system-mips: -net nic,vlan=0: 'vlan' is deprecated. Please use 'netdev' instead.
qemu-system-mips: terminating on signal 2 from pid 117071 (timeout)
Inferring network...
Interfaces: []
Done!

ddcc, 2019-12-09:
This is a known limitation of our approach, because we are replacing the original kernel with our own instrumented one. Please refer to the paper for the full details.

0xbird, 2019-12-09:
Is there any way I can solve this problem, this problem has troubled me for a long time

 
&mdash;

ddcc, 2019-12-09:
Your comment is really difficult to read. Unfortunately, you'll probably need to look into each individual firmware manually, you can refer to the FAQ for some suggestions, and also look into some of the other issues that have been opened regarding debugging with gdbserver, connecting over the serial port, or making changes to NVRAM.

0xbird, 2019-12-09:
I'm sorry, my English is not very good, thank you very much for your suggestions

ddcc, 2019-12-09:
Your english is fine, the problem is that your mail client is mangling the HTML formatting in your replies, which makes them hard to read on github.

ftk-sostupid, 2020-01-10:
> Please refer to the paper for the full details.

Could you tell me where I can find the paper?
Maybe it's because of my careless I didn't see that.

0BuRner, 2020-01-10:
@ftk-sostupid for real ?! in the readme dude... https://github.com/firmadyne/firmadyne/blob/master/paper/paper.pdf

ftk-sostupid, 2020-01-10:
> 
> 
> @ftk-sostupid for real ?! in the readme dude... https://github.com/firmadyne/firmadyne/blob/master/paper/paper.pdf

Oh, my bad, I never look down there..
So I take this id on github. lol.
Thanks.

qingguozth, 2020-04-11:
Have you solved the problem? @0xbird

0xbird, 2020-04-11:
I can't  resolved it

 

 
&mdash;