firmadyne/firmadyne #124: Wan port cable is not connected. How to emulate?

netsecu, 2019-12-11:
Hi, I did successfull emulation of different tp-link based routers like dir320 or dir615 or 740n.
In all cases I can't connect wan port - but it works, I added new tap device, I can get dhcp working.
But on router still is the message that cable is not connected. How to emulate this?

ddcc, 2019-12-11:
I'm not sure I understand the question. The emulation isn't designed to connect the emulated firmware to the internet.

netsecu, 2019-12-11:
I need to emulate wan port connection.
What I did in config: 
${QEMU} -m 256 -M ${QEMU_MACHINE} -kernel ${KERNEL} \
    -drive if=ide,format=raw,file=${IMAGE} -append "root=${QEMU_ROOTFS} console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0" \
    -nographic \
    **-net nic,vlan=0 -net tap,vlan=0,id=net0,ifname=${TAPDEV_0},script=no \
    -net nic,vlan=1 -net tap,vlan=1,id=net1,ifname=${TAPDEV_1},script=no \
    -net nic,vlan=2 -net tap,vlan=2,id=net2,ifname=${TAPDEV_2},script=no \
    -net nic,vlan=3 -net tap,vlan=3,id=net3,ifname=${TAPDEV_3},script=no** | tee ${WORK_DIR}/qemu.final.serial.log
All is working, I can see the traffic on emulated WAN port, but somehow router firmware knows that port is not up (it says "cable is not connected", which is typical for real hardware.). How to bypass this? Is this possible?

ddcc, 2019-12-11:
Sorry, I'm not sure how the firmware performs cable detection, it could be specific to the network adapter driver. You'll probably need to reverse engineer the message printing to figure out how it works.

qingguozth, 2020-04-10:
Have you solved the problem of disconnected cables?

qingguozth, 2020-04-10:
Have you solved the problem of disconnected cables? Did you succeed？

netsecu, 2020-04-10:
> Have you solved the problem of disconnected cables? Did you succeed？

unfortunately, no