firmadyne/firmadyne #125: Error after running run.sh

huynhvinhluu, 2019-12-18:
I received the notice :

Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 1000
Bringing up TAP device...
Adding route to 192.168.1.1...
Starting firmware emulation... use Ctrl-a + x to exit
qemu-system-mipsel: -nic: invalid option
Deleting route...
Bringing down TAP device...
Deleting TAP device tap1_0...
Set 'tap1_0' nonpersistent

I did not change anything in run.sh

ddcc, 2019-12-18:
Should be fixed in 5b7f4d0, update the repository and try again

huynhvinhluu, 2019-12-19:
I found the problem. Due to recent updates in scripts. I have replaced the older scripts and it worked properly

ddcc, 2019-12-19:
There was a bug earlier, but it should be fixed in the latest master, let me know if it still doesn't work.