firmadyne/firmadyne #128: A simulation problem

pedada, 2020-01-20:
When I want to emulate the DIR645 firmware, the last step when I run run.sh, it reports an error:

> qemu-system-mipsel: -netdev tap, id = net0, ifname = tap6_0, script = no: Duplicate ID 'net0' for netdev

The qemu command corresponding to run.sh is as follows:

> ${QEMU} -m 256 -M ${QEMU_MACHINE} -kernel ${KERNEL} \ 
-drive if=ide,format=raw,file=${IMAGE} -append "root=${QEMU_ROOTFS} console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel     print-fatal-signals=1 user_debug=31 firmadyne.syscall=0" \
     -nographic \                                                                 
    -netdev user,id=net0 -device e1000,netdev=net0 -nic socket,listen=:2000 -netdev user,id=net1 -device e1000,netdev=net1 -nic socket,listen=:2001 -netdev tap,id=net0,ifname=${TAPDEV_    0},script=no -device e1000,netdev=net0 -netdev user,id=net3 -device e1000,netdev=net3 -nic socket,listen=:2003 | tee ${WORK_DIR}/qemu.final.serial.log

Seems like id net0 appears twice.
DIR645 downland link: ftp://ftp2.dlink.com/PRODUCTS/DIR-645/REVA/DIR-645_FIRMWARE_1.03.ZIP

ddcc, 2020-01-20:
Sorry, it seems like some recent changes have broken things. I haven't had time to look at it yet, hopefully later this week. Does rolling back as mentioned in #127 work?

ddcc, 2020-03-01:
Should be fixed in 52fd43d81b4cbd377fc9b2ecbdc4472027701c48

closed 2020-03-01 by ddcc