firmadyne/firmadyne #135: How to access /firmadyne/ttyS1 console?

MatthewShao, 2020-03-19:
First thank you for this great tool!
I met some firmware which can boot up successfully and the web interface is accessible. But no console was spawn on the default terminal. I can manully insert `/bin/sh` into `rc` file and get the shell but the booting process will be blocked, what I want is get a shell while not blocking the booting process.

I know that `/firmadyne/console` is designed to solve this situation, but how can I connect this `firmadyne/ttyS1` character device to qemu and access another shell from the host?

ddcc, 2020-03-19:
You can try the `scripts/run-debug.sh` and see if it works. Otherwise, you could try to have `rc` run the shell in the background with `/bin/sh &`.

MatthewShao, 2020-03-20:
Thanks. In most of our samples, inserting `/bin/sh` can spwan a console but block the booting process. We've tried `/bin/sh &` but it could not spwan a console.
A possible workaround is adding `init=/bin/sh` in `-append` option of `run.sh`, which guarantees an interactive console, then we could trigger the booting process in the console by invoking something like `/etc/init.d/rcS &`. But I think maybe spwaning a non-blocking console on a new commnuication channel would be a better solution.

ddcc, 2020-03-20:
I presume you've looked at https://github.com/firmadyne/console already? It's been long enough since I touched this that I don't remember if the QEMU arguments need to be modified, and how well it worked. @megele might know more.

MatthewShao, 2020-03-26:
I figure out this by myself, actually we should add the serial devices with following options:
```sh
-serial tcp::4444,server -serial tcp::5555,server
```
this will redirect `ttyS0` to port `4444` and `ttyS1` to `5555`. With the execve hook in the kernel, we can nc to localhost:5555 and wait for a shell when it is triggered.

Notice that `virt` machine type of `armel` does not support mutiple serial port: https://unix.stackexchange.com/questions/479085/can-qemu-m-virt-on-arm-aarch64-have-multiple-serial-ttys-like-such-as-pl011-t.
We could apply the patch from the answer to newer version qemu and solve the problem.

closed 2020-03-26 by MatthewShao