firmadyne/firmadyne #136: 2 Bridge IPs but none of them is accessible

F01TH, 2020-03-27:
when i run firmadyne with Di859,i get the following message:

 iot@ubuntu  ~/source/firmware-analysis-toolkit/firmadyne  ➦ fdda7fb ●  python fat.py   

                               __           _   
                              / _|         | |  
                             | |_    __ _  | |_ 
                             |  _|  / _` | | __|
                             | |   | (_| | | |_ 
                             |_|    \__,_|  \__|                    
                    
                Welcome to the Firmware Analysis Toolkit - v0.2
    Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
                  By Attify - https://attify.com  | @attifyme
    
[?] Enter the name or absolute path of the firmware you want to analyse : DIR859Ax_FW105b03.bin
[?] Enter the brand of the firmware : Dlink
[+] Now going to extract the firmware. Hold on..
[+] Firmware : DIR859Ax_FW105b03.bin
[+] Brand : Dlink
[+] Database image ID : 105570
[+] Identifying architecture
[+] Architecture : mipseb
[+] Storing filesystem in database
[!] Filesystem already exists
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : [('br0', '192.168.0.1'), ('br1', '192.168.7.1')]
[+] Running the firmware finally
[+] command line : sudo /home/iot/source/firmware-analysis-toolkit/firmadyne/scratch/105570/run.sh
[*] Press ENTER to run the firmware...

it seems that there are two ips, but none of them is accessible. the first time simulating the firmware when i use ifconfig to check my network interface, there are two interface which are "192.168.0.2" and "192.168.7.2". but when i use ping, i get the following message :

PING 192.168.7.1 (192.168.7.1) 56(84) bytes of data.
From 192.168.7.2 icmp_seq=1 Destination Host Unreachable
From 192.168.7.2 icmp_seq=2 Destination Host Unreachable
From 192.168.7.2 icmp_seq=3 Destination Host Unreachable

and when i simulating the firmware second time, i even can not get "192.168.0.2" and "192.168.7.2" interfaces , so i want to know how to solve the problem.

pup2y, 2020-03-27:
I also have this issue!

ddcc, 2020-03-30:
Other people have reported problems with firewall or routing configuration on the firmware itself, have you checked those from a shell?

transmissionator, 2020-05-12:
Also having this same problem, tripple checked that it's nothing to do with the firewall.