firmadyne/firmadyne #137: Uable to read device config.

pup2y, 2020-03-27:
This issue will happen no matter what firmware I use.

xmldbc: can't load library '/firmadyne/libnvram.so'
devconf: '/firmadyne/libnvram.so' is not an ELF file
devconf: can't load library '/firmadyne/libnvram.so'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!     Uable to read device config.     !!
!! Setting is reset to factory default. !!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
  DEFNODE[/etc/defnodes/S11devdata.xml]
xmldbc: '/firmadyne/libnvram.so' is not an ELF file
xmldbc: can't load library '/firmadyne/libnvram.so'
  DEFNODE[/etc/defnodes/S12devdata.php]

ddcc, 2020-03-30:
That looks to be firmware specific. Have you mounted the filesystem to check if the NVRAM library is located correctly?

ppcrab, 2022-03-16:
same error...can't fix it