firmadyne/firmadyne #138: do you think that changing the mac address while emulating possible?

minanagehsalalma, 2020-03-30:
[image]

[hg531s1.zip](https://github.com/qilingframework/qiling/files/4183068/hg531s1.zip)

ddcc, 2020-03-30:
Assuming you can get access a shell, you can use nay of the normal Linux commands to change MAC addresses.

minanagehsalalma, 2020-03-30:
@ddcc Hmmm....  what if i want to see how does the def wifi password change when changing the mac address... which will be set at the very first boot of the firmware ... wouldn't the mac change be possible only after it boots?

ddcc, 2020-03-30:
In that case, you'd want to check if the kernel ethernet driver supports a customizable MAC address as a boot parameter, or just modify the kernel directly.

megele, 2020-03-30:
Any reason why you cannot use qemu's command line options to configure the mac address that the emulated nic reports up to the OS? That should work just fine for your usecase?