firmadyne/firmadyne #142: Doesn't give an Network IP

kolet, 2020-05-09:
hey there,
im not getting any Network interface ip.

i tried multiple setup.

(vmware) LTS 18.04, 20
and linux sub system.
none of them are giving an ip

i tried multiple firmware i found online .
and no one of them are giving an ip.
not sure what to do.

[image]

also tried to go there https://github.com/firmadyne/firmadyne

and follow the guide , it did not work aswell.

this Netgear WNAP320 did give an ip.
but others did not.

any guide on how to edit and what to edit so it gives an ip to other firmwares aswell ?

extremecoders-re, 2020-05-11:
Unfortunately, not all firmware images work out of the box. Manual modification may be needed on a case by case basis.

First try increasing the timeout value as mentioned in [Additional Notes](https://github.com/attify/firmware-analysis-toolkit#additional-notes).

If that doesn't work, you can check the kernel boot log to see if at all it assigns an IP to an interface. 
https://github.com/attify/firmware-analysis-toolkit/issues/43#issuecomment-607651532

kolet, 2020-05-13:
hey there. thanks for replying . @extremecoders-re 
i've read other issue.
so changing  the timeout to 5min was the first thing i've done.
it did not help .

i've also read that comment, however i did not understand what to do incase no interface is showing .