firmadyne/firmadyne #143: Problems with runExploits.py

BitnomadLive, 2020-05-18:
Hi,

in runExploits.py there are two groups of exploits. The ones you run with Metasploit and the once run with curl.

Like this one for example
curl -L --max-redir 0 -m 5 -s -f -X POST -d \"macAddress=000000000000;cat DEADBEEF5;&reginfo=1&writeData=Submit\" http://%(target)s/boardDataWW.php"
followed by
 "grep -qs \"DEADBEEF5\" qemu.serial"

So the probem I have here is that no file named qemu.serial exists. There is a log file called qemu.final.serial.log.  In your grep command you are looking for DEADBEEF5 which was part of the curl. So I assume it has to be in the log, but it isn't in the log qemu.final.serial.log.

So I was wondering how to make this work. Any help would be appreciated.

Kind regards

ddcc, 2020-05-18:
I think we turned up the kernel logging to print out all of the relevant system calls, you can see the logging configuration here: https://github.com/firmadyne/kernel-v4.1/blob/firmadyne-v4.1.17/drivers/firmadyne/hooks.c

As for the serial log, it should be qemu.final.serial.log, I think we renamed the log file at some point and forgot to update that script.

ddcc, 2020-05-18:
You could try changing the command to write into e.g. /tmp to see if the issue has to do with the root filesystem not being read-write.

That's the right setting, but it's a bitmask from 0 to 16 so you can set it to 255 to enable everything. It'll be a lot of data, but you should be able to verify that it's working by looking at the kernel log.

BitnomadLive, 2020-05-18:
I tried that but the network still doesn't get inferred. It doesn't make any sense to me how changing the output level of information is affecting the network-scripts. I read through all of the code but I can't pinpoint why firmadyne.syscall=1 doesn't cause any problems but any other value does.

BitnomadLive, 2020-05-18:
Okay so I think I found the problem inferNetwork.sh opens up makeNetwork.py and in there is QEMUCMDTEMPLATE in where I changed firmadyne.syscall to 255 but firmadyne.syscall in run.mipseb.sh has to be 1 for it to work. 
Now I am getting a IP for the firmware but the firmware  just hangs it self after a few second with this output:

[    9.544000] firmadyne: do_execve[PID: 107 (rcS)]: argv:
[   74.092000] BUG: soft lockup - CPU#0 stuck for 61s! [rcS:104]
[   74.092000] Modules linked in:
[   74.092000] Cpu 0
[   74.092000] $ 0   : 00000000 1000a400 0000006c 391aeb00
[   74.092000] $ 4   : 8f954068 8fb44e00 00000001 8f954038
[   74.092000] $ 8   : 1000a401 1000001f 8fb44e08 ffffffff
[   74.092000] $12   : ffe693b4 8f954068 00000002 8f954038
[   74.092000] $16   : 00000038 7fb5c3e0 7f9a25a4 ffffffff
[   74.092000] $20   : 7fb5c4a4 00406dac 00000002 00407b30
[   74.092000] $24   : 8fb44dd0 8067ab10                  
[   74.092000] $28   : 8fb32000 8fb33e60 7fb5b078 8011eac0
[   74.092000] Hi    : 00000002
[   74.092000] Lo    : 391aeb00
[   74.092000] epc   : 801011e0 handle_bp_int+0x18/0x58
[   74.092000]     Not tainted
[   74.092000] ra    : 8011eac0 kretprobe_trampoline+0x0/0xc
[   74.092000] Status: 1000a403    KERNEL EXL IE 
[   74.092000] Cause : 50808000
[   74.092000] PrId  : 00019300 (MIPS 24Kc)

ddcc, 2020-05-18:
Yeah the easiest way is to let it infer the network like normal, then just modify the QEMU commands in the generated run.sh script. The execve hook is known to be a bit buggy, you should turn that one off.

closed 2020-06-01 by BitnomadLive