firmadyne/firmadyne #144: Accessing the dataset of firmware

heshananupama, 2020-05-19:
Hi,
I am masters student, working in a research related to embedded firmware reverse engineering. For our work, we are in need of a dataset and after going through your papers I realized that it would be great if we could access the set of firmware that you already crawled and downloaded. Is there a way to access these? I tried going to the github repo but could not find the set of firmware that you had been used. Thank you.

ddcc, 2020-05-20:
Sorry, I don't have the dataset, and we can't redistribute it due to copyright. You can look at the provided database dump to see which firmware were tested, and run the scraper to crawl and download firmware.

heshananupama, 2020-05-20:
Thanks Dominic. Is this the schema under the database folder? I could not locate a dump other than the schema

ddcc, 2020-05-20:
No, it is this link in the README: https://github.com/firmadyne/firmadyne#data

heshananupama, 2020-05-20:
Thank you Dominic!

closed 2020-05-20 by heshananupama