firmadyne/firmadyne #147: cannot find __inet_insert_ifa in qemu.initial.serial.log

jackulder, 2020-07-12:
Hey there!
I'm trying to emulate DIR-601_REVB_FIRMWARE_2.01.BIN. First I followed the instructions from [here](https://github.com/attify/firmware-analysis-toolkit) and I got these:
`[+] Firmware: DIR-601_REVB_FIRMWARE_2.01.BIN`
`[+] Extracting the firmware...`
`[+] Image ID: 1`
`[+] Identifying architecture...`
`[+] Architecture: mipseb`
`[+] Building QEMU disk image...`
`[+] Setting up the network connection, please standby...`
`[+] Network interfaces: []`
`[+] All set! Press ENTER to run the firmware...`
`[+] When running, press Ctrl + A X to terminate qemu`

then I read [https://github.com/attify/firmware-analysis-toolkit/issues/43#issuecomment-607651532](https://github.com/attify/firmware-analysis-toolkit/issues/43#issuecomment-607651532) and understood that there is no "__inet_insert_ifa" in qemu.initial.serial.log as shown below:

```[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.17+ (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:05:21 UTC 2016
[    0.000000] CPU: ARMv7 Processor [412fc0f1] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine model: linux,dummy-virt
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c061dfe8, node_mem_map cfdf9000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: firmadyne.syscall=1 root=/dev/vda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 253340K/262144K available (4297K kernel code, 170K rwdata, 1584K rodata, 180K init, 148K bss, 8804K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc05c67bc   (5882 kB)
[    0.000000]       .init : 0xc05c7000 - 0xc05f4000   ( 180 kB)
[    0.000000]       .data : 0xc05f4000 - 0xc061e840   ( 171 kB)
[    0.000000]        .bss : 0xc0621000 - 0xc06462d4   ( 149 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 62.50MHz (virt).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x1cd42e208c, max_idle_ns: 881590405314 ns
[    0.000086] sched_clock: 56 bits at 62MHz, resolution 16ns, wraps every 4398046511096ns
[    0.000164] Switching to timer-based delay loop, resolution 16ns
[    0.002179] Console: colour dummy device 80x30
[    0.002418] Calibrating delay loop (skipped), value calculated using timer frequency.. 125.00 BogoMIPS (lpj=625000)
[    0.002506] pid_max: default: 32768 minimum: 301
[    0.003338] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.003364] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.008215] CPU: Testing write buffer coherency: ok
[    0.014064] Setting up static identity map for 0x40008240 - 0x40008298
[    0.024371] VFP support v0.3: implementor 41 architecture 4 part 30 variant f rev 0
[    0.030591] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.089650] NET: Registered protocol family 16
[    0.091772] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.094618] cpuidle: using governor ladder
[    0.094715] cpuidle: using governor menu
[    0.109824] Serial: AMBA PL011 UART driver
[    0.111357] 9000000.pl011: ttyS0 at MMIO 0x9000000 (irq = 53, base_baud = 0) is a PL011 rev1
[    0.122304] console [ttyS0] enabled
[... 151 lines cut ...]
[    0.998260] NET: Registered protocol family 10
[    1.003258] ip6_tables: (C) 2000-2006 Netfilter Core Team
[    1.004760] sit: IPv6 over IPv4 tunneling driver
[    1.007580] NET: Registered protocol family 17
[    1.008129] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    1.008458] Bridge firewalling registered
[    1.008655] Ebtables v2.0 registered
[    1.009406] 8021q: 802.1Q VLAN Support v1.8
[    1.009894] Registering SWP/SWPB emulation handler
[    1.011964] rtc-pl031 9010000.pl031: setting system clock to 2020-07-12 10:04:01 UTC (1594548241)
[    1.019468] EXT4-fs (vda1): couldn't mount as ext3 due to feature incompatibilities
[    1.020562] EXT4-fs (vda1): mounting ext2 file system using the ext4 subsystem
[    1.029440] EXT4-fs (vda1): mounted filesystem without journal. Opts: (null)
[    1.085966] VFS: Mounted root (ext2 filesystem) on device 254:1.
[    1.091088] Freeing unused kernel memory: 180K (c05c7000 - c05f4000)
[    1.099664] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
[    1.099920] CPU: 0 PID: 1 Comm: swapper Tainted: G        W       4.1.17+ #1
[    1.100051] Hardware name: Generic DT based system
[    1.100170] [<c001c99c>] (unwind_backtrace) from [<c0019d30>] (show_stack+0x10/0x14)
[    1.100311] [<c0019d30>] (show_stack) from [<c0410b78>] (panic+0x70/0x1c8)
[    1.100497] [<c0410b78>] (panic) from [<c040f33c>] (kernel_init+0xb8/0xe4)
[    1.100619] [<c040f33c>] (kernel_init) from [<c0016da8>] (ret_from_fork+0x14/0x2c)
[    1.100865] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.```

Does anyone have any idea?

ddcc, 2020-07-13:
I've added a FAQ entry regarding the init process. The problem is that it has an unusual name, so it isn't being executed after the system boots. You'll need to manually identify the correct one, and specify it as an additional kernel boot parameter.

jackulder, 2020-07-14:
> I've added a FAQ entry regarding the init process. The problem is that it has an unusual name, so it isn't being executed after the system boots. You'll need to manually identify the correct one, and specify it as an additional kernel boot parameter.

Thank you. I read that, but still I'm not sure how to solve it. Could you explain more about it and which script should I modify?

ddcc, 2020-07-14:
You'll need to use the mount and unmount scripts to look inside the filesystem. You can search online and read the Linux init.txt file mentioned in the error for some background on what the init process does. Then, after you identify the correct one, you append the boot parameter that specifies the init path to the QEMU kernel boot parameters. For your firmware, it's the long quotes string following `-append` in `run.mipseb.sh`. You'll probably also need to add it to `run.sh` later as well, assuming that the network inference works correctly and generates that file.