firmadyne/firmadyne #152: Why I use qemu-system-mips in DECAF, the network doesn't work and has this error "NETDEV WATCHDOG: eth0 (pcnet32): transmit queue 0 timed out"

yinfeidi, 2020-09-12:
Please press Enter to activate this console. [   54.832000] ------------[ cut here ]------------
[   54.832000] WARNING: at net/sched/sch_generic.c:255 dev_watchdog+0x160/0x2c4()
[   54.832000] NETDEV WATCHDOG: eth0 (pcnet32): transmit queue 0 timed out
[   54.832000] Modules linked in:
[   54.832000] Call Trace:
[   54.832000] [<8056b7fc>] dump_stack+0x8/0x34
[   54.832000] [<80124180>] warn_slowpath_common+0x78/0xa4
[   54.832000] [<80124234>] warn_slowpath_fmt+0x2c/0x38
[   54.832000] [<8044a5e4>] dev_watchdog+0x160/0x2c4
[   54.832000] [<801302a4>] run_timer_softirq+0x188/0x254
[   54.832000] [<80129f68>] __do_softirq+0xcc/0x190
[   54.832000] [<8012a1c8>] do_softirq+0x5c/0x94
[   54.832000] [<8012a414>] irq_exit+0x4c/0xbc
[   54.832000] [<80101a44>] ret_from_irq+0x0/0x4
[   54.832000] [<80101c80>] r4k_wait+0x20/0x40
[   54.832000] [<801037a4>] cpu_idle+0x30/0x60
[   54.832000] [<80706908>] start_kernel+0x3bc/0x3dc
[   54.832000] 
[   54.832000] ---[ end trace 1e517aac3df85c47 ]---
[   54.832000] pcnet32: eth0: transmit timed out, status 07fb, resetting
^C
Please press Enter to activate this console.

ddcc, 2020-09-12:
I have no idea what DECAF is. I also don't see anything obviously wrong with the network, seems like a QEMU configuration issue?

yinfeidi, 2020-09-12:
I checked the QEMU version used in DECAF and found it is 1.0, so it may be the reason?
-------------------------------------------------------------
$ ./qemu-system-mips-decaf --version

(process:8109): GLib-WARNING **: 16:58:37.637: ../../../../glib/gmem.c:489: custom memory allocation vtable not supported
QEMU emulator version 1.0, Copyright (c) 2003-2008 Fabrice Bellard

ddcc, 2020-09-12:
Yeah that's pretty old. Have you tried with a newer version of QEMU?

yinfeidi, 2020-09-16:
Yeah, but DECAF is hard to use. So do you have some good suggestions to trace the QEMU running firmware. As I want to detect the inside behavior in the firmware.

ddcc, 2020-09-16:
No idea what DECAF is still. QEMU lets you enable a GDB stub that can debug the running kernel. Alternatively, you can inject/patch instrumentation for whatever binaries you'd like on the system, and have them print things to standard output.