firmadyne/firmadyne #153: Can you please make firmadyne kali compatible.

csenger18, 2020-09-12:
Distro details:

PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2020.3"
VERSION_ID="2020.3"
VERSION_CODENAME="kali-rolling"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"
-------------------------------------------------------

Kernel version: 5.7.0-kali3-amd64

ddcc, 2020-09-12:
Sorry, firmadyne is not under active development, but we'll review and accept pull requests

csenger18, 2020-09-15:
I booted up an Ubuntu 14.04.6 LTS live disk and get started to do the checklist, then I get this message: "E: Package uml-utilities has no installation candidate", but after I pasted these in sources list

deb http://archive.ubuntu.com/ubuntu/ trusty main restricted universe multiverse 
deb http://archive.ubuntu.com/ubuntu/ trusty-security main restricted universe multiverse 
deb http://archive.ubuntu.com/ubuntu/ trusty-updates main restricted universe multiverse 
deb http://archive.ubuntu.com/ubuntu/ trusty-proposed main restricted universe multiverse 
deb http://archive.ubuntu.com/ubuntu/ trusty-backports main restricted universe multiverse 

is seems to be working, after installation comes the next problem: jefferson have some pylzma problem, but after 
git cloned manually and installed, this solved the problem.  

please Modify this line in README.md  "5. sudo -H pip install git+https://github.com/sviehb/jefferson."  to manual install  

In this vid the guy installs some packages that might be needed for firmadyne: https://www.youtube.com/watch?v=U86iFH7muwg

ddcc, 2020-09-15:
Not sure why you're using such an old version of Ubuntu, the problems you encountered are probably related.

csenger18, 2020-09-15:
"Setup
The following has been tested on a Ubuntu 14.04 machine."

ddcc, 2020-09-15:
Oh, I should remove that, it's from 5 years ago.

csenger18, 2020-09-15:
Firmadyne support armv7?? If detects armel can be changed to ARM???

csenger18, 2020-09-15:
Any advice for compatible OS versions??

ddcc, 2020-09-15:
Not sure what you're asking. ARM uses the virt machine platform in QEMU, which defaults to a ARMv7-A Cortex-A15 processor. armel just refers to little-endian ARM, as opposed to big-endian.

Just use whatever modern distribution, although as a developer tool, the assumption is that you can fix any issues that arise.

csenger18, 2020-09-15:
Ah, thanks for the info 👍

csenger18, 2020-09-15:
Lastly I found the real problem in the deps.sh (binwak) libqt4-opengl, Python-lzo, Python-lzma

Somehow can be used the  installed version of binwalk not the git cloned.

csenger18, 2020-09-15:
Above mentioned depencies I only found in the Ubuntu repository.

csenger18, 2020-09-16:
Compatibilty problem with the latest Ubuntu 20.04.1 LTS (up to date fresh)
tester@tester-VirtualBox:~$ sudo apt-get install busybox-static fakeroot git dmsetup kpartx netcat-openbsd nmap python-psycopg2 python3-psycopg2 snmp uml-utilities util-linux vlan
Reading package lists... Done
Building dependency tree       
Reading state information... Done
busybox-static is already the newest version (1:1.30.1-4ubuntu6.1).
busybox-static set to manually installed.
dmsetup is already the newest version (2:1.02.167-1ubuntu1).
dmsetup set to manually installed.
fakeroot is already the newest version (1.24-1).
fakeroot set to manually installed.
netcat-openbsd is already the newest version (1.206-1ubuntu1).
netcat-openbsd set to manually installed.
util-linux is already the newest version (2.34-0.1ubuntu9.1).
util-linux set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 nmap : Depends: python:any
 python-psycopg2 : Depends: python (>= 2.7) but it is not going to be installed
                   Depends: python (< 2.8) but it is not going to be installed
                   Depends: python:any (>= 2.7.1-0ubuntu2)
                   Recommends: python-egenix-mxdatetime but it is not going to be installed
 python3-psycopg2 : Depends: python3 (< 3.5) but 3.8.2-0ubuntu2 is to be installed
 snmp : Depends: libsnmp30 (>= 5.7.2~dfsg) but it is not going to be installed
        Depends: libsnmp-base but it is not going to be installed
 uml-utilities : Depends: libreadline6 (>= 6.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

csenger18, 2020-09-16:
The last  compatible version is Ubuntu 16.04??  or have any newer compatible version?

ddcc, 2020-09-16:
Did you update your package information first? Not sure why package dependencies aren't being installed. With 20.04, python2 has been completely deprecated, so the python2-psycopg2 package isn't available anymore. I haven't tested python3 compatibility, so I don't know if it'll make a difference.

csenger18, 2020-09-18:
installation try on ubuntu 20.04.1
[20.04.zip](https://github.com/firmadyne/firmadyne/files/5244482/20.04.zip)

I also have good news: I installed the Ubuntu 18.04.5 LTS next to my Kali (successfully running on Ubuntu  the firmadyne)

and also working with  firmware-analysis-toolkit  (https://github.com/attify/firmware-analysis-toolkit)

please write in the README:"The following has been tested on a Ubuntu 18.04.5 LTS"

some repository might be need from: trusty

csenger18, 2020-09-18:
Some details about the D-link DIR-825 C1's firmware running with firmadyne: 
1, get the same ip with my home router and no GUI on the web interface.
2, when it's running it's messed up my internet connection untill I stop the emulation.
[DIR825C1_FW300B32.zip](https://github.com/firmadyne/firmadyne/files/5244777/DIR825C1_FW300B32.zip)

ddcc, 2020-09-18:
That's not how you install missing packages. You can't just add old repositories because package dependencies may change, and will probably break your system. Aside from the migration away from python2, the issue is that you haven't enabled the repositories that contain additional software, e.g. universe.

As for networking, if there's an IP conflict, then that's expected behavior and not really a problem with firmadyne. The solution is to either manually move one network into a non-conflicting range, or manually configure your routing table to prefer one interface and explicitly specify routes for the other.

csenger18, 2020-09-19:
I reinstalled my Ubutu 18.04.1 LTS based on your advices ( enables  repo,  not adding old)
now firmadyne seems ok :)

Can I ask for a sample firmware like Netgear WNAP320 but with armel architecture?

csenger18, 2020-09-19:
Is it solvable in firmadyne?
[image]

csenger18, 2020-09-19:
Is it possible to run those  devices firmware which one does not have default ip? (the user have to setup after the device started)

ddcc, 2020-09-19:
I don't know of any offhand for armel.

You can look at the FAQ entry, but otherwise it's probably not going to work without manual effort for reversing.