firmadyne/firmadyne #159: Not everything is using Firmadyne's libnvram.so via LD_PRELOAD

amlamarra, 2020-10-06:
If I understand correctly, Firmadyne's kernel ensures that the init process uses Firmadyne's libnvram.so library for LD_PRELOAD in the file "init/main.c" at line 153:
`char * envp_init[MAX_INIT_ENVS+3] = { "HOME=/", "TERM=linux", "LD_PRELOAD=/firmadyne/libnvram.so", NULL, };`
It also seems like any services started from init will use that LD_PRELOAD as well, correct?

The problem I'm running into is when I start services or run programs manually and they rely on nvram, but don't use Firmadyne's libnvram. So I usually see a "/dev/nvram: No such device" error message, like:
```
# nvram get http_passwd
/dev/nvram: No such device
```

I can replace the device's `libnvram.so` library with Firmadyne's version, which usually works. However, then the device won't boot properly. I'm guessing there's some functions in there that aren't implemented in Firmadyne's version. OR I can manually specify LD_PRELOAD:
```
# LD_PRELOAD=/firmadyne/libnvram.so nvram get http_passwd
nvram_get_buf: http_passwd = "admin"
admin
```

Is there a way to set a global LD_PRELOAD? Does it depend on the device? One device I'm working with does not have an "/etc/profile" file for setting environment variables.
For the one that does have that file, I've tried adding `export LD_PRELOAD=/firmadyne/libnvram.so`. However, that variable doesn't get added to the environment vars in the console shell.

ddcc, 2020-10-07:
Yeah, the idea is that everything should inherit the default environment, but it's a bit buggy and doesn't seem to always work. A better solution is probably to modify the execve syscall to ensure the preload always included in the environment.

I don't know of an easier way to do this in userspace, because devices will use different shells and C runtime libraries, which is probably why the environment isn't always being inherited.

cq674350529, 2020-10-09:
There is another way to specifiy the preload library in userspace: using `/etc/ld.so.preload`.  I came across some cases when the `LD_PRELOAD` env didn't work, but using the `/etc/ld.so.preload` works for me.  Maybe you can have a try @amlamarra

amlamarra, 2020-10-10:
@cq674350529 Thank you, I never knew about that. I created the `/etc/ld.so.preload` file with the contents `/firmadyne/libnvram.so`. However, I still get the same results (`/dev/nvram: No such device`).

It seems like that's something used by glibc and not necessarily uclibc. I did see an `/etc/ld.so.conf` file on this device with the contents:
```
/lib
/usr/lib
```

So I appended the line `/firmadyne` do the beginning of that file. Still get the same results.

ddcc, 2020-10-10:
Sounds like the issue may not be with the preload, but rather that the functions aren't been intercepted. Have you verified that the NVRAM function called is one that the firmadyne library intercepts?

amlamarra, 2020-10-11:
> Sounds like the issue may not be with the preload, but rather that the functions aren't been intercepted. Have you verified that the NVRAM function called is one that the firmadyne library intercepts?

I have verified that the interception works when I use LD_PRELOAD:
```
# LD_PRELOAD=/firmadyne/libnvram.so nvram get http_passwd
nvram_get_buf: http_passwd = "admin"
admin
```

ddcc, 2020-10-12:
Can you modify the init or other startup scripts to add LD_PRELOAD? Otherwise you could modify the kernel to insert it into every execve system call.

amlamarra, 2020-10-12:
There are no init/startup scripts. It's all controlled by the `rc` binary. And of course the source for that was not included in the GPL release.

I didn't know I could modify the kernel to insert it into every execve system call. I'll have to look into that.

amlamarra, 2021-05-26:
This is no longer needed.

closed 2021-05-26 by amlamarra