firmadyne/firmadyne #160: Mounting error in emulation

Galla218, 2020-10-07:
Attempting to emulate firmware for a security camera (ARM) on Ubuntu 16. Running into this problem:

`cat qemu-final-serial.log`

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.17+ (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:05:21 UTC 2016
[    0.000000] CPU: ARMv7 Processor [412fc0f1] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine model: linux,dummy-virt
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c061dfe8, node_mem_map cfdf9000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: root=/dev/vda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 253340K/262144K available (4297K kernel code, 170K rwdata, 1584K rodata, 180K init, 148K bss, 8804K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc05c67bc   (5882 kB)
[    0.000000]       .init : 0xc05c7000 - 0xc05f4000   ( 180 kB)
[    0.000000]       .data : 0xc05f4000 - 0xc061e840   ( 171 kB)
[    0.000000]        .bss : 0xc0621000 - 0xc06462d4   ( 149 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 62.50MHz (virt).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x1cd42e208c, max_idle_ns: 881590405314 ns
[    0.000141] sched_clock: 56 bits at 62MHz, resolution 16ns, wraps every 4398046511096ns
[    0.000282] Switching to timer-based delay loop, resolution 16ns
[    0.003077] Console: colour dummy device 80x30
[    0.003390] Calibrating delay loop (skipped), value calculated using timer frequency.. 125.00 BogoMIPS (lpj=625000)
[    0.003499] pid_max: default: 32768 minimum: 301
[    0.004364] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.004400] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.015386] CPU: Testing write buffer coherency: ok
[    0.025973] Setting up static identity map for 0x40008240 - 0x40008298
[    0.042785] VFP support v0.3: implementor 41 architecture 4 part 30 variant f rev 0
[    0.055939] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.066486] NET: Registered protocol family 16
[    0.069321] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.078839] cpuidle: using governor ladder
[    0.078989] cpuidle: using governor menu
[    0.119578] Serial: AMBA PL011 UART driver
[    0.121564] 9000000.pl011: ttyS0 at MMIO 0x9000000 (irq = 53, base_baud = 0) is a PL011 rev1
[    0.131000] console [ttyS0] enabled
[    0.179329] vgaarb: loaded
[    0.180885] SCSI subsystem initialized
[    0.182004] usbcore: registered new interface driver usbfs
[    0.182398] usbcore: registered new interface driver hub
[    0.182753] usbcore: registered new device driver usb
[    0.192876] cfg80211: Calling CRDA to update world regulatory domain
[    0.196233] Switched to clocksource arch_sys_counter
[    0.219678] NET: Registered protocol family 2
[    0.225542] TCP established hash table entries: 2048 (order: 1, 8192 bytes)
[    0.225862] TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
[... 160 lines cut ...]
[    1.996371] 8021q: adding VLAN 0 to HW filter on device eth0
mount: mounting /dev/mtdblock4 on /mnt/mtd/ipc failed: Invalid argument

can't open /dev/ttySGK0: No such file or directory
/etc/init.d/S90ipc: line 5: /mnt/mtd/ipc/run: not found

can't open /dev/ttySGK0: No such file or directory
[    3.346795] cfg80211: Calling CRDA to update world regulatory domain

can't open /dev/ttySGK0: No such file or directory

can't open /dev/ttySGK0: No such file or directory

can't open /dev/ttySGK0: No such file or directory
[    6.503832] cfg80211: Calling CRDA to update world regulatory domain

can't open /dev/ttySGK0: No such file or directory

can't open /dev/ttySGK0: No such file or directory
```
It fails on this line in `S90ipc` `mount -t squashfs /dev/mtdblock4 /mnt/mtd/ipc`. The next line is another mount `mount -t jffs2 /dev/mtdblock5 /mnt/mtd/ipc/conf` and appears to mount successfully because no error message is returned and instead errors on the next line `/mnt/mtd/ipc/run &`. Is it not recognizing the squashfs type? 

As for the `can't open /dev/ttySGK0: No such file or directory` I'm guessing I either need to add this into the firmware or is related to the above problems? I'm a novice at this so I apologize if the answer is obvious.

Any insights you might have into this issue are greatly appreciated.

ddcc, 2020-10-07:
Do you see any of the /dev/mtdblock* devices inside the emulated firmware? They probably don't exist, so they will fail to mount.

Similarly, there's probably no /dev/ttySGK0 device either. In order to add a dummy one, you'll need to modify the kernel/drivers/firamdyne/devfs_stubs.c file to add a line for that device, then recompile the kernel.

Galla218, 2020-10-08:
The mtdblock* devices do exist inside the emulated firmware. Mounting the image I can see them in /dev

```
ubuntu:~/firmadyne/scratch/1/image$ ls dev/

adsl0    hidraw0  mtd   mtd10  mtd4  mtd7  mtdblock   mtdblock10  mtdblock4  mtdblock7  mtdr0   mtdr2  mtdr5  mtdr8  ppp     tts   ttyS0  ttyS3
armem    kmem     mtd0  mtd2   mtd5  mtd8  mtdblock0  mtdblock2   mtdblock5  mtdblock8  mtdr1   mtdr3  mtdr6  mtdr9  ptmx    tty   ttyS1  urandom
console  mem      mtd1  mtd3   mtd6  mtd9  mtdblock1  mtdblock3   mtdblock6  mtdblock9  mtdr10  mtdr4  mtdr7  null   random  tty0  ttyS2  zero
```

ddcc, 2020-10-08:
In that case, the emulated mtdblock device is likely uninitialized or empty, so it won't be recognized as containing a valid squashfs image. Does the boot process actually pause or stop if an error occurs? You could try to e.g. substitute the mount with a ramdisk, load an empty squashfs filesystem into the mtdblock before it is mounted, or just comment out the mount and see if the boot process continues.

Galla218, 2020-10-09:
Commenting out the mount doesn't stop the boot process. Still stalls on `can't open /dev/ttySGK0: No such file or directory`. I've tried rebuilding the kernel and adding in a line for the device but am not sure what the second and third args should be. Doing a `grep -R` for SGK0 showed in `etc/inittab` the following lines:

```
# Put a getty on the serial port
ttySGK0::respawn:/sbin/getty -L ttySGK0 115200 ttySGK0 # GENERIC_SERIAL
```
So I added  `DEVICE(ttySGK0, 4, 64, open, read, write, close, ioctl)` but had no luck.

ddcc, 2020-10-09:
The second and third argument should be the [major and minor numbers](https://www.oreilly.com/library/view/linux-device-drivers/0596000081/ch03s02.html) for the device file. If the ttySGK0 file exists on the filesystem, you can use ls to show them. However, even if the numbers are incorrect, the device file should still be able to be opened.

Can you try just making a regular empty file at /dev/ttySGK0, and booting with the original unmodified kernel? Does that make a difference?

Galla218, 2020-10-09:
> The second and third argument should be the [major and minor numbers](https://www.oreilly.com/library/view/linux-device-drivers/0596000081/ch03s02.html) for the device file.

That's what I thought and the few I've tried haven't worked. Adding in an empty file doesn't work either still continues with saying there is no file/directory. I appreciate all of your help but think I'm going to look for a different approach now.

closed 2020-10-09 by Galla218