firmadyne/firmadyne #163: kernel message to report author

santhoshn16, 2020-10-19:
[image]

ddcc, 2020-10-19:
Thanks for the report, but I think this is an issue with the firmware rather than the kernel itself.

santhoshn16, 2020-10-20:
okay, If can you help me in emulating a firmware, I have attached the qemu logs for that firmware. I just want to know in which components of the firmware are having trouble. 
[qemu.final.serial.log](https://github.com/firmadyne/firmadyne/files/5407961/qemu.final.serial.log)
[qemu.initial.serial.log](https://github.com/firmadyne/firmadyne/files/5407962/qemu.initial.serial.log)

ddcc, 2020-10-20:
There isn't any obvious. You'll probably need to get your hands dirty to figure out what the boot process is doing, where the ioctl errors are coming from, and whether they are related or not. There's an FAQ entry on debugging in the README.