firmadyne/firmadyne #168: [Question] Is there some sort of caching mechanism in place for each run?

kr1tzy, 2020-11-06:
I'm trying to get to the bottom of an issue I'm having every time I run Firmadyne with a modified `libnvram.so`. I'm using the included `libnvram/` source to build out a new shared object file and copy it into `scratch/1/image/firmadyne/libnvram.so`. It runs as expected after 3(ish) attempts but acts as if the previous `libnvram.so` is still in place the first few runs. 

I perused the source to see if there was a flag or something to disable a cache but only came across one mention of cache in the [db schema](https://github.com/firmadyne/firmadyne/search?q=cache). 

Rebuilding the image works but that's basically just as inconvenient as running it three times to test my new `libnvram.so`.

Any feedback is greatly appreciated.

ddcc, 2020-11-07:
The default makeImage.sh script will inject libnvram.so from the download pre-built binaries in the binaries/ subfolder. So if you don't overwrite those, and re-execute the script, your compiled version will probably be overwritten.

kr1tzy, 2020-11-10:
@ddcc, replacing the libnvram.so.mipsel (in my case) in `binaries/` and then running  the `makeImage.sh` script works as expected; it's when I mount the root filesystem with `mount.sh` and replace the `scratch/<index>/image/firmadyne/libnvram.so` directly that the behavior is non-deterministic. It uses the previous `libnvram.so` for 2-3 runs and then eventually uses the one I replaced it with. That's why i was curious about some sort of caching mechanism. If it's not obvious from your stand point it might be a qemu or host specific thing. Either way, I appreciate the feedback and quick response.

ddcc, 2020-11-10:
Are you sure that you're cleanly mounting and unmounting the filesystem, with no remaining instances of QEMU still accessing the disk? There shouldn't be any other caching mechanism.

kr1tzy, 2020-11-17:
I haven't checked but will next time i run it and report back. This can be closed.

closed 2020-11-17 by kr1tzy