firmadyne/firmadyne #170: Syncing disks. ----Mounting QEMU Image---- /dev/mapper/control: open failed: Operation not permitted Failure to communicate with kernel device-mapper driver. Check that device-mapper is available in the kernel. device mapper prerequisites not met

benbenlala, 2020-11-13:
I buil it in docker container.when i run ./scripts/makeImage.sh 1,
some error occurs.

Syncing disks.
----Mounting QEMU Image----
/dev/mapper/control: open failed: Operation not permitted
Failure to communicate with kernel device-mapper driver.
Check that device-mapper is available in the kernel.
device mapper prerequisites not met

ddcc, 2020-11-13:
Sounds like a Docker issue. You'll probably need to configure it to grant appropriate system call permissions to the container.

oztechmuse, 2021-07-07:
devicemapper is deprecated in Docker and doesn't seem to work (at least on Mac) - they have moved to overlay2 - so not a permissions issue - it would be good if someone could look at this as not able to progress running firmadyne in Docker

AndrewFasano, 2021-09-19:
If you have qemu-nbd loaded on your *host* kernel, you should be able to get the docker container working if you start the container with the extra flags `--privileged-v /dev:/dev`