firmadyne/firmadyne #174: Not network

azpema, 2021-01-29:
Hello, I am trying to use firmadyne with a router firmware, but it is unable to infer the network. I have read several other issues related to this but these do not seem similar to my problem. The following is the initial log:

`[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:39:21 UTC 2016
[    0.000000] 
[    0.000000] LINUX started...
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0061e000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f8f1000 @ 0070e000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 57792 bytes for tracking 1806 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x0000ffff
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x0000ffff
[    0.000000] On node 0 totalpages: 65535
[    0.000000] free_area_init_node: node 0, pgdat 806aa3c0, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60959 pages, LIFO batch:15
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65023
[    0.000000] Kernel command line: firmadyne.syscall=1 root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rw debug ignore_loglevel print-fatal-signals=1 init=/sbin/init
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252428k/254916k available (4260k kernel code, 2252k reserved, 1549k data, 220k init, 0k highmem)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:256
[    0.000000] CPU frequency 200.00 MHz
[    0.000000] Console: colour dummy device 80x25
[    0.004000] Calibrating delay loop... 968.70 BogoMIPS (lpj=1937408)
[    0.088000] Mount-cache hash table entries: 512
[    0.112000] NET: Registered protocol family 16
[    0.120000] bio: create slab <bio-0> at 0
[    0.124000] vgaarb: loaded
[    0.124000] SCSI subsystem initialized
[    0.124000] libata version 3.00 loaded.
[    0.124000] usbcore: registered new interface driver usbfs
[    0.128000] usbcore: registered new interface driver hub
[    0.128000] usbcore: registered new device driver usb
[    0.128000] pci 0000:00:00.0: reg 14 32bit mmio pref: [0x1000000-0x1ffffff]
[    0.132000] pci 0000:00:0a.1: reg 20 io port: [0x00-0x0f]
[    0.132000] pci 0000:00:0a.2: reg 20 io port: [0x00-0x1f]
[    0.132000] pci 0000:00:0a.3: BAR 8: address space collision on of bridge [0x1100-0x110f]
[    0.132000] pci 0000:00:0a.3: quirk: region 1100-110f claimed by PIIX4 SMB
[    0.132000] pci 0000:00:12.0: reg 10 32bit mmio pref: [0x000000-0x1ffffff]
[    0.132000] pci 0000:00:12.0: reg 14 32bit mmio: [0x000000-0x000fff]
[    0.132000] pci 0000:00:12.0: reg 30 32bit mmio pref: [0x000000-0x00ffff]
[    0.136000] pci 0000:00:13.0: reg 10 32bit mmio: [0x000000-0x01ffff]
[    0.136000] pci 0000:00:13.0: reg 14 io port: [0x00-0x3f]
[    0.136000] pci 0000:00:13.0: reg 30 32bit mmio pref: [0x000000-0x07ffff]
[    0.136000] pci 0000:00:14.0: reg 10 32bit mmio: [0x000000-0x01ffff]
[    0.136000] pci 0000:00:14.0: reg 14 io port: [0x00-0x3f]
[    0.136000] pci 0000:00:14.0: reg 30 32bit mmio pref: [0x000000-0x07ffff]
[    0.136000] pci 0000:00:15.0: reg 10 32bit mmio: [0x000000-0x01ffff]
[    0.136000] pci 0000:00:15.0: reg 14 io port: [0x00-0x3f]
[    0.136000] pci 0000:00:15.0: reg 30 32bit mmio pref: [0x000000-0x07ffff]
[... 337 lines cut ...]
Enter 'help' for a list of built-in commands.

# cat: can't open '/var/bdmf_sh_id': No such file or directory
Invalid session_id -cmd after -c or too few parameters
Usage:
	bdmf_shell server_addr:port [options]
	bdmf_shell domain_socket_addr [options]
	bdmf_shell [-c init | session_id] [-cmd command] [-f script_file] [-fg] [-close]
		-c : serial communication. Must be followed by "init" or session_id
		-cmd command : send BDMF shell command
		-f script_file : send script_file content
		-fg - interactive mode
		-close - close session opened by "-c init" - used with "-c"
[   51.144000] firmadyne: sys_socket[PID: 251 (ifconfig)]: family:2, type:1, protocol:0
ifconfig: SIOCSIFADDR: No such device
/etc/extra_func.sh: line 16: dropbearkey: not found
`

As shown in the above log it only generates the loop interface:
`[    8.956000] firmadyne: __inet_insert_ifa[PID: 77 (ifconfig)]: device:lo ifa:0x7f000001
[    8.964000] firmadyne: __inet_insert_ifa[PID: 77 (ifconfig)]: device:lo ifa:0x7f000001
[    9.004000] firmadyne: __inet_insert_ifa[PID: 77 (ifconfig)]: device:lo ifa:0x7f000001`

What could the issue be?
Thanks

ddcc, 2021-01-31:
Did you look through the log messages? Seems like the boot process quit because the SPI flash is missing, so you'll need to figure out a way around that.

megele, 2021-02-01:
additionally, it seems that the ssk and smd processes die. These immediately (in the log) precede ethctl which complains that eth0/eth1 are missing. Might be worthwhile trying to figure out what's going on with these too.