firmadyne/firmadyne #176: Failure to emulate OpenWRT based firmware - Possibly due to partition layout - TP-Link Archer C7 v5

gdynamics, 2021-03-09:
I am trying to emulate an Archer C7 1750 v5's firmware using Firmadyne.

From everything I can tell from OpenWRT themselves (https://openwrt.org/toh/tp-link/archer-c7-1750) and TP-Link themselves (https://www.tp-link.com/us/support/download/archer-c7/), this router's firmware is based on OpenWRT 19.07.7. 

It does not look like they made any large, severe changes to the operating system itself from what I can tell, but grain of salt given this is my first time getting moderately deep in embedded systems.

---

I assume I am as far as can be from the first person to have tried to emulate OpenWRT or a TP-Link router on Firmadyne, but I see no other issues open or closed relating to OpenWRT or TP-Link which mirror the issues I have below. This may be a known issue that other people have figured out much more easily than me, but I am unaware of such a solution.

I have two guesses as to the RCA of my issues, but I am unsure how to proceed.

1. Could it be **incompatibility with OpenWRT's partition layout** (https://openwrt.org/docs/techref/flash.layout)? It appears on some level like OpenWRT has quite a complicated partition layout, and that scripts cannot find important directories because they are on a partition which has not been mounted, as it does not exist within Firmadyne's single ext2 partition layout defined in `makeImage.sh`.
2. Could it be **preInit.sh not being run**? Firmadyne being unable to mount partitions that OpenWRT needs to run seems like a definite possible cause, considering the boot messages.

---

All issues below can be reproduced through the firmware available at https://www.tp-link.com/us/support/download/archer-c7/, and simply following steps 1 through 8 in the generic Firmadyne `README.md`.

---

**The first issue** I ran into was that the preInit.sh script seems to not even run. After attempting to add prints & infinite loops in the beginning of preInit.sh, nothing changed.

_My change to preInit.sh_
[image]

I tested both at the image creation & with manual image mounting to confirm that this updated script was in the image, and QEMU appears to point to the correct place.

_run.sh & preInit.sh_
[image]

---

**The second issue** appears to be OpenWRT initialization scripts not actually being able to mount anything.

_Boot failures_
[image]

The main scripts are `/sbin/mount_root` and `/lib/functions/boot.sh`, where `/sbin/mount_root` calls functions from `/lib/functions/boot.sh`

_/sbin/mount_root calling functions from /lib/functions/boot.sh_
[image]
_Some contents of /lib/functions/bootsh_
[image]

---

**The third issue** I ran into was that run.sh never drops a terminal/shell on any interface after any length of time. By modifying `/etc/preinit`, I was able to `ls /dev/tty*` and discovered ttyS0 through 3. 

_Modified /etc/preinit & its results during a boot_
[image]

However, specifying ttyS1-3 through the `run.sh` QEMU `console` argument did not print out any additional information after initial boot, while the default ttyS0 continues to print errors.

I also attempted to force busybox to drop an ash shell through `/etc/preinit` as well, but it did not work, but I am not very confident in my methodology.

---

I apologize for the physically large post, but I am hoping by showing all my issues & useful information through screenshots it will help with understanding.

ddcc, 2021-03-10:
Thanks for all the information! I don't have a lot of time to help with debugging, but I'd suggest starting with the first issue. 

There's probably a bug here in the firmadyne scripts, because according to the kernel documentation, `rdinit` is only valid for a ramdisk filesystem, but currently that script is only included in the root filesystem. It's been a long time since I've worked on this, and the ramdisk may have gotten lost in the process of upgrading kernel versions. As a workaround, you should be able to set the `init` boot parameter to run the preinit.sh script as init instead, and then modify the preinit.sh script to call the real init.

gdynamics, 2021-03-14:
You are certainly right about `rdinit` being the reason it does not run `preInit.sh`, and simply changed it from `rdinit` to `init` as you suggested.

For documentation purposes, I was able to bypass an issue with `/run` not existing by just modding `preInit.sh` to create it before mounting it.

At the end of my `preInit.sh` I call `/etc/preinit` directly, but somewhere along the way it fails silently, causing a kernel panic.
[image]

After doing some grep'ing around for `- init-`, I found it in `/lib/preinit/99_10_run_init`, and added some prints to trace execution. I found that it was failing on the `exec env` line, and decided to force it to go the path which does not suppress stderr, and print out the contents of the execution and found that it was executing `/sbin/init`.
Using the documentation about the OpenWRT boot process from https://openwrt.org/docs/techref/process.boot, once execution is passed to `/sbin/init`, here are the steps:
[image]

 I decided to instrument the script that is called by init, `/etc/init.d/rcS`, since the `/etc/inittab` file is present and points to the right place. However, after adding prints to said script, I noticed that it never prints.
[image]

I expect that `/sbin/init` would be able to execute and do as the OpenWRT documentation states. However, the lack of error output from the `exec env` in `99_10_run_init` and the lack of printing from `rcS` has me stumped. If init has the location of the initialization script and the script exists, why would it not run the script?

ddcc, 2021-03-15:
Are you sure the output of `rcS` will actually be printed to console, rather than logged into a file or something?

gdynamics, 2021-03-22:
I believe it should because of these:
https://serverfault.com/questions/341919/how-to-find-error-messages-from-linux-init-d-rc-d-scripts
https://unix.stackexchange.com/questions/56075/why-is-rcs-required-after-file-system-is-mounted-by-the-kernel

I attempted to add a test to the beginning rcS as well to have it simply create a file, and that does not work.
[image]

ddcc, 2021-03-23:
Sounds like /sbin/init is crashing somewhere during the boot process, before it gets to rcS then. You can try to enable more of the kernel print hooks (see https://github.com/firmadyne/kernel-v2.6#introduction), by appending the "firmadyne.syscall=255" parameter to the QEMU kernel parameters.

gdynamics, 2021-03-26:
It looks like you're right, but I am trying to think of where to go next.

Immediately after the call to `/sbin/init`
[image]

...a bunch of libraries being loaded...

The panic
[image]

Is it even starting `/sbin/init` correctly? We don't get any syscalls after the libraries and fs ioctl's

ddcc, 2021-03-26:
Looks to me like init is exiting for some reason after the ioctl. You can try to debug this either statically or dynamically; statically by disassembling the binary or otherwise following the source code to figure out why it might be exiting, or dynamically by replacing init with e.g. bash, and attempting to strace or otherwise debug init. For the latter, you may need to compile statically-linked versions of strace, gdb, or other debugging tools though.

ddcc, 2021-05-10:
Since you're doing a static build, it doesn't really matter what the C library running on the target is. So I'd suggest just spinning up e.g. a Debian MIPS VM, and compile a static version of strace on there. That way, you can avoid issues with cross-compile toolchains and potential libc incompatibilities. If the latest version of strace doesn't work, you may need to try an older version though, since the firmadyne kernel is fairly old.