firmadyne/firmadyne #179: Extractor problems

combat-entropy, 2021-05-03:
I am trying and failing to extract the Archer VR600v v170814 firmware which is downloadable from https://static.tp-link.com/Archer_VR600v(EU)_V2_170814.zip .
I can extract the example WNAP320 firmware so I believe I have configured firmadyne correctly. 

Initially I believe I was hitting a recursion limit but I increased the RECURSION_BREADTH RECURSION_DEPTH in extractor.py without it helping. 

Strangely calling binwalk -Mre directly it works and extracts files. 

When running ./sources/extractor/extractor.py -b TPLink -sql 127.0.0.1 -np -nk "Archer_VR600v(EU)_V2_170814.zip" images 
I get:
```
>> Database Image ID: 6

/home/ce/firmadyne/Archer_VR600v(EU)_V2_170814.zip
>> MD5: 3adbe70bb4d48cd3fcf3a2be5006cac5
>> Tag: 6
>> Temp: /tmp/tmpj8prb8d9
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>>>> Zip archive data, at least v2.0 to extract, compressed size: 28694085, uncompressed size: 33854588, name: Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin
>> Recursing into archive ...
	
/tmp/tmpj8prb8d9/_Archer_VR600v(EU)_V2_170814.zip.extracted/Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin
	>> MD5: d61687d93c1bb9990f3f04b89b8d388d
	>> Tag: 6
	>> Temp: /tmp/tmpo3t7bs8c
	>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
	>> Recursing into archive ...
	>>>> JFFS2 filesystem, big endian
	>>>> Zlib compressed data, best compression
	>> Recursing into compressed ...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/1184DA5
		>> MD5: 1d3fda2edb4a89ab60a23c5f7c7d81dd
		>> Tag: 6
		>> Temp: /tmp/tmp6dce2_a6
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmp6dce2_a6...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/118C12E
		>> MD5: 90a888263cd3c7763e191e14d7c87bc9
		>> Tag: 6
		>> Temp: /tmp/tmpk1pikxqz
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmpk1pikxqz...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/118D223
		>> MD5: e76466ff28e7b22d8b4e4e62ba05ba65
		>> Tag: 6
		>> Temp: /tmp/tmp4xg3678q
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmp4xg3678q...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/118D485
		>> MD5: 1d3fda2edb4a89ab60a23c5f7c7d81dd
		>> Skipping: 1d3fda2edb4a89ab60a23c5f7c7d81dd...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/11CD676
		>> MD5: 29ebceec041869772bc09410872dbcfa
		>> Tag: 6
		>> Temp: /tmp/tmp5o0rp3mw
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmp5o0rp3mw...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/11CDB25
		>> MD5: 1d3fda2edb4a89ab60a23c5f7c7d81dd
		>> Skipping: 1d3fda2edb4a89ab60a23c5f7c7d81dd...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/11EF656
		>> MD5: 2e5369671b5325eb00368f519e566ad6
		>> Tag: 6
		>> Temp: /tmp/tmpg_jqo9oh
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmpg_jqo9oh...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/11F108D
		>> MD5: 1d3fda2edb4a89ab60a23c5f7c7d81dd
		>> Skipping: 1d3fda2edb4a89ab60a23c5f7c7d81dd...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin-0.extracted/region.json
		>> MD5: 0bcd451073b19a10c66f668d0bf643e6
		>> Tag: 6
		>> Temp: /tmp/tmp8a6cloyc
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmp8a6cloyc...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin.extracted/jffs2-root/fs_1/cferam.000
		>> MD5: 7fc82e6d097f02016ec80375afb056b1
		>> Tag: 6
		>> Temp: /tmp/tmp0b5yxsj2
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmp0b5yxsj2...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin.extracted/jffs2-root/fs_1/tag_kernel
		>> MD5: 9e4ef7d7a7215fcfdc56b28229d210d7
		>> Tag: 6
		>> Temp: /tmp/tmpuvm66scx
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>> Recursing into compressed ...
		>> Cleaning up /tmp/tmpuvm66scx...
		
/tmp/tmpo3t7bs8c/_Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin.extracted/jffs2-root/fs_1/vmlinux.lz
		>> MD5: 275f0ddaba503d18195520024f50c8e8
		>> Tag: 6
		>> Temp: /tmp/tmpmvqhw3ah
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>> Recursing into archive ...
		>>>> LZMA compressed data, properties: 0x6D, dictionary size: 4194304 bytes, uncompressed size: 5776624 bytes
		>> Recursing into compressed ...
			
/tmp/tmpmvqhw3ah/_vmlinux.lz.extracted/C
			>> MD5: 14c8f0204452627c1f202f8c63f8c8ce
			>> Tag: 6
			>> Temp: /tmp/tmpphn8v29o
			>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
			>> Recursing into archive ...
			>>>> gzip compressed data, maximum compression, from Unix, last modified: 1970-01-01 00:00:00 (null date)
			>> Recursing into compressed ...
				
/tmp/tmpphn8v29o/_C.extracted/43D668
				>> MD5: e84fc59c49aa2d640ea956aac715c3c8
				>> Skipping: text/plain...
			>> Cleaning up /tmp/tmpphn8v29o...
		>> Cleaning up /tmp/tmpmvqhw3ah...
	>> Cleaning up /tmp/tmpo3t7bs8c...
>> Recursing into compressed ...
	
/tmp/tmpj8prb8d9/_Archer_VR600v(EU)_V2_170814.zip.extracted/Archer_VR600vV2_1.0.0_0.9.1_up_boot(170814)_2017-08-14_18.28.17.bin
	>> MD5: d61687d93c1bb9990f3f04b89b8d388d
	>> Skipping: d61687d93c1bb9990f3f04b89b8d388d...
>> Cleaning up /tmp/tmpj8prb8d9...

```

ddcc, 2021-05-06:
Looks like the extracted JFFS2 filesystem isn't being recognized as a valid UNIX-like filesystem. Are you sure the extracted filesystem is actually complete and appears to be bootable? There's a UNIX_THRESHOLD variable that requires at least four of UNIX_DIRS to be treated as a valid filesystem.

combat-entropy, 2021-05-06:
I've tried setting UNIX_THRESHOLD as low as 1 with no improvement. 

Binwalk output when scanning the firmware:
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
132096        0x20400         JFFS2 filesystem, big endian
2884608       0x2C0400        UBI erase count header, version: 1, EC: 0x0, VID header offset: 0x800, data offset: 0x1000

The extracted filesystem looks complete to me but I am not an expert. I have attached the tree output of what binwalk extracted from the firmware. 
[binwalk-tree.txt](https://github.com/firmadyne/firmadyne/files/6437170/binwalk-tree.txt)

It seems like firmadyne is perhaps ignoring the UBIFS filesystem.

ddcc, 2021-05-06:
Oh, I think I see the problem. Can you confirm that manually launching `binwalk -y filesystem` on the input also misses the UBI result? I suspect that because the keyword filesystem is not present in that binwalk result, it is being skipped by the binwalk filtering and the extractor. If that is the case, can you also check if `binwalk -y filesystem -y UBI` fixes it by including that entry?

ddcc, 2021-05-06:
Also, as a workaround, you can manually extract the root filesystem into a compressed file name `<image id>.tar.gz`, and place that into the images directory.

combat-entropy, 2021-05-06:
binwalk -y filesystem Archer_VR600vV2_1.0.0_0.9.1_up_boot\(170814\)_2017-08-14_18.28.17.bin 

```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
132096        0x20400         JFFS2 filesystem, big endian
3150848       0x301400        UBIFS filesystem superblock node, CRC: 0x62786DA8, flags: 0x0, min I/O unit size: 2048, erase block size: 126976, erase block count: 231, max erase blocks: 2048, format version: 4, compression type: zlib
3281920       0x321400        UBIFS filesystem master node, CRC: 0x15CC8787, highest inode: 1172, commit number: 0
3412992       0x341400        UBIFS filesystem master node, CRC: 0x19FC729A, highest inode: 1172, commit number: 0
```
binwalk -y filesystem -y UBI gives the same output.

combat-entropy, 2021-05-06:
Sorry didn't notice your other post. I did create the tar.gz manually and was getting "Kernel panic - not syncing: No init found." error in qemu, I thought I may have created my tar,gz incorrectly and hence created this issue so I might do it "properly" and hopefully have no init error to troubleshoot.

ddcc, 2021-05-07:
Seems to be an issue with how binwalk signatures for UBIFS are defined. Notice the difference between the following two executions:

```
$ binwalk -y ubi Archer_VR600vV2_1.0.0_0.9.1_up_boot\(170814\)_2017-08-14_18.28.17.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
2884608       0x2C0400        UBI erase count header, version: 1, EC: 0x0, VID header offset: 0x800, data offset: 0x1000

$ binwalk -y filesystem Archer_VR600vV2_1.0.0_0.9.1_up_boot\(170814\)_2017-08-14_18.28.17.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
132096        0x20400         JFFS2 filesystem, big endian
3150848       0x301400        UBIFS filesystem superblock node, CRC: 0x62786DA8, flags: 0x0, min I/O unit size: 2048, erase block size: 126976, erase block count: 231, max erase blocks: 2048, format version: 4, compression type: zlib
3281920       0x321400        UBIFS filesystem master node, CRC: 0x15CC8787, highest inode: 1172, commit number: 0
3412992       0x341400        UBIFS filesystem master node, CRC: 0x19FC729A, highest inode: 1172, commit number: 0
```

Although both detect UBIFS signatures, only the first matches at a valid offset that is an extractable UBIFS filesystem; the three results from the latter one all not extractable.

ddcc, 2021-05-07:
Turns out `binwalk -y filesystem -y ubi` is the correct work-around, I just had the capitalization wrong earlier.

commit 45b57b5a (2021-05-07): bump extractor, fix #179
closed 2021-05-07 by ddcc