firmadyne/firmadyne #181: Unable to extract filesystem from Netgear R6700v3 firmware

amlamarra, 2021-05-26:
I first did this with the latest [R6700v1 firmware](https://www.downloads.netgear.com/files/GDC/R6700/R6700-V1.0.2.16_10.0.57.zip) and it worked fine. But for some reason, the latest [R6700v3 firmware](https://www.downloads.netgear.com/files/GDC/R6700v3/R6700v3-V1.0.4.106_10.0.80.zip) does not. However, `binwalk -e` is able to extract the filesystem just fine. In fact, it extracts the same filesystem twice:
```plaintext
$ binwalk -e R6700v3-V1.0.4.106_10.0.80.chk

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
58            0x3A            TRX firmware header, little endian, image size: 47251456 bytes, CRC32: 0xA5F4D05B, flags: 0x0, version: 1, header size: 28 bytes, loader offset: 0x1C, linux kernel offset: 0x20BD9C, rootfs offset: 0x0
86            0x56            LZMA compressed data, properties: 0x5D, dictionary size: 65536 bytes, uncompressed size: 5276608 bytes
2145750       0x20BDD6        Squashfs filesystem, little endian, version 4.0, compression:xz, size: 45101299 bytes, 1813 inodes, blocksize: 131072 bytes, created: 2020-11-25 01:21:33

$ ls _R6700v3-V1.0.4.106_10.0.80.chk.extracted/
20BDD6.squashfs  56  56.7z  squashfs-root  squashfs-root-0

$ du -sh _R6700v3-V1.0.4.106_10.0.80.chk.extracted/*
44M	_R6700v3-V1.0.4.106_10.0.80.chk.extracted/20BDD6.squashfs
5.1M	_R6700v3-V1.0.4.106_10.0.80.chk.extracted/56
46M	_R6700v3-V1.0.4.106_10.0.80.chk.extracted/56.7z
150M	_R6700v3-V1.0.4.106_10.0.80.chk.extracted/squashfs-root
150M	_R6700v3-V1.0.4.106_10.0.80.chk.extracted/squashfs-root-0
```

So the first time I did this, it was taking a long time and I just walked away from it. I tried it again and realized that this was filling up the hard drive in the /tmp directory. I guess it's recursively extracting that `20BDD6.squashfs` file.

Here's the output from Firmadyne from my first attempt:
```plaintext
$ ./sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk R6700v3-V1.0.4.106_10.0.80.zip images
>> Database Image ID: 2

/home/andrew/firmadyne/R6700v3-V1.0.4.106_10.0.80.zip
>> MD5: fde021eba9a2a26f5947fba60da489b7
>> Tag: 2
>> Temp: /tmp/tmpfaq8bqzn
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>>>> Zip archive data, at least v1.0 to extract, compressed size: 47251514, uncompressed size: 47251514, name: R6700v3-V1.0.4.106_10.0.80.chk
>> Recursing into archive ...
	
/tmp/tmpfaq8bqzn/_R6700v3-V1.0.4.106_10.0.80.zip.extracted/R6700v3-V1.0.4.106_10.0.80.chk
	>> MD5: 043a62421dc7d9efa43590c2f8d8d764
	>> Tag: 2
	>> Temp: /tmp/tmp6vvdojpu
	>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
	>>>> Zip archive data, at least v1.0 to extract, name: dap_data/
	>> Recursing into archive ...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk.extracted/dap_data/dap
		>> MD5: 23dd2ee230f68ffe208bec10e0c498c5
		>> Skipping: application/x-executable...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk.extracted/dap_data/nginx
		>> MD5: 80db4d364a3788fe059769a914fbf809
		>> Skipping: application/x-executable...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk.extracted/dap_data/dap_logd
		>> MD5: 2aad70b6421a84cabb416bfcfd98b5be
		>> Skipping: application/x-executable...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk.extracted/dap_data/dap_daemon
		>> MD5: 7e56ab1b3c8fe3e4d0e79bf93544e1fe
		>> Skipping: application/x-executable...
	>>>> Squashfs filesystem, little endian, version 4.0, compression:xz, size: 45101299 bytes, 1813 inodes, blocksize: 131072 bytes, created: 2020-11-25 01:21:33
	>>>> LZMA compressed data, properties: 0x5D, dictionary size: 65536 bytes, uncompressed size: 5276608 bytes
	>> Recursing into compressed ...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk-1.extracted/56
		>> MD5: dc4e70809478882f27ed99e4f62de86d
		>> Tag: 2
		>> Temp: /tmp/tmpu3sug2nv
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>>>> ASCII cpio archive (SVR4 with no CRC), file name: "/dev", file name length: "0x00000005", file size: "0x00000000"
		>> Recursing into archive ...
		>>>> LZMA compressed data, properties: 0xC0, dictionary size: 0 bytes, uncompressed size: 64 bytes
		>> Recursing into compressed ...
			
/tmp/tmpu3sug2nv/_56-0.extracted/43F5B
			>> MD5: 3b5d3c7d207e37dceeedd301e35e2e58
			>> Tag: 2
			>> Temp: /tmp/tmpp4qjfho_
			>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
			>> Recursing into archive ...
			>> Recursing into compressed ...
			>> Cleaning up /tmp/tmpp4qjfho_...
			
/tmp/tmpu3sug2nv/_56-0.extracted/4EA877
			>> MD5: 70bc8f4b72a86921468bf8e8441dce51
			>> Tag: 2
			>> Temp: /tmp/tmpkk7qu7aw
			>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
			>> Recursing into archive ...
			>> Recursing into compressed ...
			>> Cleaning up /tmp/tmpkk7qu7aw...
		>> Cleaning up /tmp/tmpu3sug2nv...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk-1.extracted/20BE36
		>> MD5: 1e14215758b12d13bfd5ee55c4c1df3e
		>> Skipping: application/x-executable...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk-1.extracted/21A2DE
		>> MD5: 3a12244374a7a32f025d78db51dad37d
		>> Tag: 2
		>> Temp: /tmp/tmpap85wte3
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>>>> PARity archive data - file number 18768
		>> Recursing into archive ...
		>>>> bzip2 compressed data, block size = 100k
		>> Recursing into compressed ...
			
/tmp/tmpap85wte3/_21A2DE.extracted/2DBE2
			>> MD5: 65ae24e3e54451c242863768ddbda328
			>> Tag: 2
			>> Temp: /tmp/tmpuz4faxib
			>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
			>> Recursing into archive ...
			>> Recursing into compressed ...
			>> Cleaning up /tmp/tmpuz4faxib...
		>> Cleaning up /tmp/tmpap85wte3...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk-1.extracted/228EB2
		>> MD5: 63c92b89501c5d1f2772701ea34f02b3
		>> Tag: 2
		>> Temp: /tmp/tmp9ay2ag0v
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>>>> PARity archive data - file number 18768
		>> Recursing into archive ...
		>>>> bzip2 compressed data, block size = 100k
		>> Recursing into compressed ...
			
/tmp/tmp9ay2ag0v/_228EB2.extracted/DBE2
			>> MD5: 65ae24e3e54451c242863768ddbda328
			>> Skipping: 65ae24e3e54451c242863768ddbda328...
		>> Cleaning up /tmp/tmp9ay2ag0v...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk-1.extracted/236352
		>> MD5: 4aa2591cec353a94947efc087cc1f1af
		>> Skipping: application/x-executable...
		
/tmp/tmp6vvdojpu/_R6700v3-V1.0.4.106_10.0.80.chk-1.extracted/2440A6
		>> MD5: 173512ce54be51c83285fd5666f4368f
		>> Tag: 2
		>> Temp: /tmp/tmpq2bdky1u
		>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
		>>>> PARity archive data - file number 18768
		>> Recursing into archive ...
		>> Recursing into compressed ...
	>> Skipping: recursion breadth 5
		>> Cleaning up /tmp/tmpq2bdky1u...
	>> Skipping: completed!
	>> Cleaning up /tmp/tmp6vvdojpu...
	
/tmp/tmpfaq8bqzn/_R6700v3-V1.0.4.106_10.0.80.zip.extracted/R6700v3-V1.0.4.106_10.0.80_Release_Notes.html
	>> MD5: c614264e49c2792ad2f58d42dfff8581
	>> Skipping: text/html...
>>>> Squashfs filesystem, little endian, version 4.0, compression:xz, size: 45101299 bytes, 1813 inodes, blocksize: 131072 bytes, created: 2020-11-25 01:21:33
>>>> LZMA compressed data, properties: 0x5D, dictionary size: 65536 bytes, uncompressed size: 5276608 bytes
>> Recursing into compressed ...
	
/tmp/tmpfaq8bqzn/_R6700v3-V1.0.4.106_10.0.80.zip.extracted/R6700v3-V1.0.4.106_10.0.80.chk
	>> MD5: 043a62421dc7d9efa43590c2f8d8d764
	>> Skipping: 043a62421dc7d9efa43590c2f8d8d764...
	
/tmp/tmpfaq8bqzn/_R6700v3-V1.0.4.106_10.0.80.zip.extracted/R6700v3-V1.0.4.106_10.0.80_Release_Notes.html
	>> MD5: c614264e49c2792ad2f58d42dfff8581
	>> Skipping: c614264e49c2792ad2f58d42dfff8581...
Traceback (most recent call last):
  File "./sources/extractor/extractor.py", line 427, in extract
  File "./sources/extractor/extractor.py", line 640, in _check_compressed
  File "./sources/extractor/extractor.py", line 704, in _check_recursive
  File "./sources/extractor/extractor.py", line 234, in __init__
  File "./sources/extractor/extractor.py", line 114, in io_md5
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpfaq8bqzn/_R6700v3-V1.0.4.106_10.0.80.zip-0.extracted/squashfs-root-0/var'
>> Cleaning up /tmp/tmpfaq8bqzn...
```

amlamarra, 2021-05-26:
Is it easy to do this part manually?

ddcc, 2021-05-26:
Are you able to get binwalk to extract the root filesystem correctly? The first output shows it extracting various SquashFS bits, but I don't see a full filesystem (e.g. /dev, /etc/, etc.). You can always work around the extractor by manually extracting and compressing the into the tar.gz file.

amlamarra, 2021-05-26:
Yes, binwalk extracts the filesystem correctly:
```plaintext
$ ls _R6700v3-V1.0.4.106_10.0.80.chk.extracted/squashfs-root*
_R6700v3-V1.0.4.106_10.0.80.chk.extracted/squashfs-root:
bin  data  dev  etc  lib  media  mnt  opt  proc  sbin  share  sys  tmp  usr  var  www

_R6700v3-V1.0.4.106_10.0.80.chk.extracted/squashfs-root-0:
bin  data  dev  etc  lib  media  mnt  opt  proc  sbin  share  sys  tmp  usr  var  www
```
All of the expected binaries are in there. I'll try doing it manually.

It looks like the `image` table got updated:
```plaintext
❯ psql -h localhost -d firmware -U firmadyne -W -c "select * from image;"
Password: 
 id |            filename            | description | brand_id |               hash               | rootfs_extracted | kernel_extracted | arch  | kernel_version 
----+--------------------------------+-------------+----------+----------------------------------+------------------+------------------+-------+----------------
  1 | R6700-V1.0.2.16_10.0.57.zip    |             |        1 | f14adae5a0af596cd3f8c702c0ce6d70 | t                | f                | armel | 
  2 | R6700v3-V1.0.4.106_10.0.80.zip |             |        1 | fde021eba9a2a26f5947fba60da489b7 | f                | f                |       | 
```

amlamarra, 2021-05-26:
Ok, I was able to do this manually.

ddcc, 2021-06-01:
I'll leave this open as a reminder to look into the extractor at some point...

tripp-kyler, 2022-02-18:
@amlamarra , stuck in the same spot with an R8000, any chance you could elaborate on what you did next? It seems like you compressed the _R6700v3-V1.0.4.106_10.0.80.chk.extracted/squashfs-root* directory and then ran it through again but when I did this it paused while setting up the network connection, please standby...

amlamarra, 2022-02-22:
IIRC, I just extracted the filesystem with binwalk manually and put it in the images directory.