firmadyne/firmadyne #182: Unable to log into the console

wfwerf, 2021-06-17:
After emulating the firmware I am never able to get into the console with the default creds. What might be the reason to it ?
[image]

ddcc, 2021-06-18:
This isn't the example firmware, is it? You'll probably need to look at /etc/passwd or /etc/shadow to try to figure out what the credentials should be for this firmware.

wfwerf, 2021-06-18:
No its not. I took your advice and found that according to the passwd file and the shadow file. username and password are admin and sohoadmin but even that doesnot let me inside the device. Any reason why this could be ?