firmadyne/firmadyne #183: Questions on running results of scripts for analysis.

Calistamu, 2021-06-30:
Hello!
I am trying use those analysing scripts to confirm vulnerability scanning functions of this tool. And Here are some results and questions.   
(Before my analysing work, I successfully accessed this router's web page whose ip address is 192.168.0.1. This router is dir-850l.)
1. After I run ```./analyses/snmpwalk.sh 192.168.0.1```, there is no error message.But when I check snmp，I get ```Timeout: No Response from 192.168.0.1```. Am I wrong? Or I really cannot get snmp information?
[image]

2. After I run ```./analyses/webAccess.py 1 192.168.0.1 log.txt```,I got this result and I cannot understand this result.
[image]

3.Because I cannot get codes of shell exploits, I cannot the meaning of those results.
[image]

cq674350529, 2021-07-01:
Just based on the contents of related scripts (`snmpwalk.sh`, `webAccess.py` ):

> After I run ./analyses/snmpwalk.sh 192.168.0.1, there is no error message.But when I check snmp，I get Timeout: No Response from 192.168.0.1. Am I wrong? Or I really cannot get snmp information?

If the device doesn't support snmp functionality or the snmp service is disabled, then you will get no response from the device.

> After I run ./analyses/webAccess.py 1 192.168.0.1 log.txt,I got this result and I cannot understand this result.

According to the paper, this script is used for iterating through each file within the firmware image that appears to be served by a webserver (e.g., located within /www/), verifies that it is not a static resource (e.g., *.png, *.css, *.js), and attempts to access it directly over the web interface.
That is, if the requested file can't be directly accessed via `http://<ip>/<file>`, then the expected response will be `HTTP Error: 404` (file not found).

ddcc, 2021-07-02:
Yup, I've been traveling, thanks for the response!