firmadyne/firmadyne #187: No route to host

megyltv, 2021-11-03:
Hello, 
I'm trying to run FAT with the example provided under "Usage" in a virtual machine with Ubuntu 20.04. When I run a network interface is given `Network interfaces: [('eth0', '192.168.1.1')]`, but when I tried to use the browser with that IP I received this error:
[image]

In another terminal I'm mounting the image (`./scripts/mount.sh 1`) and running one of the analysis scripts (Eg. `/analysis/webAccess.py 192.168.1.1`) the following error is displayed: `-> URLError: [Errno 113] No route to host`
[image]

Is there any network interface configuration missing that needs to be added so it can display the results?
Thanks

ddcc, 2021-11-04:
Were there any issues with starting the VM using the script? It's supposed to modify the kernel routing table to add a route to the VM.

megyltv, 2021-11-10:
After troubleshooting, it looks like the emulation of the router setting the services stopped. After starting network and rcpd services, some of the issues got resolved.

Currently, the scripts, like `webAccess.py` are throwing 404 Not Found, as, for instance, `http://192.168.1.1/tmpl/BackupSettings.tpl.php` doesn't exists in the router. is that expected?

ddcc, 2021-11-11:
That's probably normal, it's just guessing URLs based off of the filesystem path, so it's more of an exploration tool than anything else.