firmadyne/firmadyne #191: Resolve Dependency and Installation Issues?

lastlistener, 2022-02-01:
Attempting to build Firmadyne from the Docker file, or just install Firmadyne on a modern OS, even after significant modification, often results in a pretty bad dependency tangle. For example, some fun I just encountered:

- Python 3.7 is required to support numerous Binwalk dependencies. Old versions of Debian/Ubuntu generally don't support Python 3.7 out of the box.
- For modern versions of Ubuntu or Debian, Binwalk is largely incompatible with those versions and Binwalk's "deps.sh" script fails silently, leaving the user to triage. This is unfriendly at best.

It would be in the world's best interest for the authors to take some corrective action, such as settling on fixed version numbers for dependencies or regularly maintaining a Dockerfile with a known good configuration. 

I'm an engineer, so I obviously _might_ be able to fix these issues myself in an isolated environment after enough digging; and I know this project was originally an academic proof of concept and not intended to be a supported software in the long term. That being said, the project has outgrown those roots and become one of the most important pieces of security tooling in the 21st century. I think it would be ideal if people could at least install it easily, without having to audit half the installed toolchain to manually resolve accumulated dependency bugs.

ddcc, 2022-02-02:
Thanks for the feedback, unfortunately none of us are able to work actively on firmadyne. Are you referring to python2.7 or python3.7? Most of our scripts should be compatible with python3, so I don't believe python2.7 is explicitly required. If it works fine, feel free to submit a PR to update it.

flamecopper, 2022-02-23:
I have tried a few options:
Firmadyne on Debian
Firmadyne on WSL on Ubuntu
FAT on WSL on Ubuntu
FAT on Ubuntu,

So far the only options that don't give me issues FAT on Ubuntu.
Firmadyne on Debian doesn't work well with deps.sh and the gcc compilation
WSL doesn't work well with postgresSQL

ByridianBlack, 2023-07-12:
> I have tried a few options:
> 
> Firmadyne on Debian
> 
> Firmadyne on WSL on Ubuntu
> 
> FAT on WSL on Ubuntu
> 
> FAT on Ubuntu,
> 
> 
> 
> So far the only options that don't give me issues FAT on Ubuntu.
> 
> Firmadyne on Debian doesn't work well with deps.sh and the gcc compilation
> 
> WSL doesn't work well with postgresSQL

Hey I am having problems installing as well. Can you let me know what you did to resolve ?