firmadyne/firmadyne #196: No network interface

anesic-c4, 2022-03-31:
All commands works fine, I even start firmware, but I never get Interface and IP address for emulated firmware. 
I used example_analysis.sh also, but no interface is running neither.
There is no any particular error. 
This is when command is ran. 

sudo ./scripts/inferNetwork.sh 2 mipsel
Running firmware 2: terminating after 60 secs...
qemu-system-mipsel: terminating on signal 2 from pid 32122 (timeout)
Inferring network...
Interfaces: []
Done!

Thanks in advance.

closed 2022-03-31 by anesic-c4

HAHALOSAH, 2022-04-28:
I'm also having this problem, how do I fix it?

ThomEsgate, 2023-02-15:
I'm also having this problem

yejuns, 2023-04-10:
I'm also having this problem, i have tried to config network, but it didn't work