firmadyne/firmadyne #197: extracotr.py needs to be run as root with the latest binwalk

amlamarra, 2022-05-01:
It seems the latest version of binwalk to needs to be run as root when being used in `extractor.py`:

```plaintext
~/firmadyne ❯ ./sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk "WNAP320 Firmware Version 2.0.3.zip" images
>> Database Image ID: 1

/home/andrew/firmadyne/WNAP320 Firmware Version 2.0.3.zip
>> MD5: 51eddc7046d77a752ca4b39fbda50aff
>> Tag: 1
>> Temp: /tmp/tmpuzbfvda9
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True

Extractor Exception: In order to execute third party applications as root, binwalk must be run with root privileges.
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/binwalk-2.3.3-py3.8.egg/binwalk/core/module.py", line 258, in __init__
    self.load()
  File "/usr/local/lib/python3.8/dist-packages/binwalk-2.3.3-py3.8.egg/binwalk/modules/extractor.py", line 160, in load
    raise ModuleException("In order to execute third party applications as %s, binwalk must be run with root privileges." % self.runas_user)
binwalk.core.exceptions.ModuleException: In order to execute third party applications as root, binwalk must be run with root privileges.
----------------------------------------------------------------------------------------------------

Traceback (most recent call last):
  File "./sources/extractor/extractor.py", line 427, in extract
  File "./sources/extractor/extractor.py", line 474, in _check_archive
  File "./sources/extractor/extractor.py", line 652, in _check_recursive
  File "/usr/local/lib/python3.8/dist-packages/binwalk-2.3.3-py3.8.egg/binwalk/__init__.py", line 10, in scan
    objs = m.execute()
  File "/usr/local/lib/python3.8/dist-packages/binwalk-2.3.3-py3.8.egg/binwalk/core/module.py", line 783, in execute
    obj = self.run(module)
  File "/usr/local/lib/python3.8/dist-packages/binwalk-2.3.3-py3.8.egg/binwalk/core/module.py", line 802, in run
    obj = self.load(module, kwargs)
  File "/usr/local/lib/python3.8/dist-packages/binwalk-2.3.3-py3.8.egg/binwalk/core/module.py", line 833, in load
    argv.update(self.dependencies(module, argv['enabled']))
  File "/usr/local/lib/python3.8/dist-packages/binwalk-2.3.3-py3.8.egg/binwalk/core/module.py", line 865, in dependencies
    raise ModuleException("Failed to load " + dependency.name + " module")
binwalk.core.exceptions.ModuleException: Failed to load Extractor module
>> Cleaning up /tmp/tmpuzbfvda9...
```

It works fine when I run that same command as root. Could that cause problems elsewhere? Or can the instructions just be updated?

ddcc, 2022-05-01:
You can try using fakeroot, I think that's how we ran the experiments originally.

amlamarra, 2022-09-18:
Sorry for taking so long. Yes, fakeroot does work.

closed 2022-09-18 by amlamarra