firmadyne/firmadyne #199: Erorr file inferNetwork.sh and erorr message: "swRegRead: Operation not supported" when run file run.sh

Huynhngockhanh52, 2022-09-12:
I have a problem with the TL841N_V14 TPlink firmware emulation. When I run the inferNetwork.sh file, I get an error like this:
[image]
`**──(root㉿kali)-[/home/kali/Downloads/firmadyne]
└─#** ./scripts/inferNetwork.sh 1      
Querying database for architecture... Password for user firmadyne: 
mipsel
Running firmware 1: terminating after 60 secs...
qemu-system-mipsel: terminating on signal 2 from pid 7520 (timeout)
Inferring network...
Interfaces: [('br0', '192.168.0.1')]
Done!
`
However, firmadyne still generates the run.sh file, when I run run.sh I get the same error that keeps repeating:
`Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 0
Initializing VLAN...
Bringing up TAP device...
Adding route to 192.168.0.1...
Starting firmware emulation... use Ctrl-a + x to exit
[    0.000000] Linux version 2.6.39.4+ (ddcc@ddcc-virtual) (gcc version 5.3.0 (GCC) ) #2 Tue Sep 1 18:11:28 EDT 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0065e000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f8b2000 @ 0074e000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 59840 bytes for tracking 1870 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x00010000
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x00010000
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 806e8800, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60960 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252372k/254664k available (4453k kernel code, 2292k reserved, 1606k data, 240k init, 0k highmem)
[    0.000000] NR_IRQS:256
[    0.000000] CPU frequency 333.35 MHz
[    0.000000] Console: colour dummy device 80x25
[    0.012000] Calibrating delay loop... 884.73 BogoMIPS (lpj=1769472)
[    0.092000] pid_max: default: 32768 minimum: 301
[    0.092000] Mount-cache hash table entries: 512
[    0.120000] Performance counters: No available PMU.
[    0.148000] NET: Registered protocol family 16
[    0.184000] bio: create slab <bio-0> at 0
[    0.192000] vgaarb: loaded
[    0.196000] SCSI subsystem initialized
[    0.196000] libata version 3.00 loaded.
[    0.200000] usbcore: registered new interface driver usbfs
[    0.200000] usbcore: registered new interface driver hub
[... 508 lines cut ...]
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
swRegRead: Operation not supported
`
My English is not good!
Please help me with this problem! Thank you very much!

juliencohenscali, 2022-12-15:
Got the same error with : [this tpLink Firmware](https://static.tp-link.com/upload/firmware/2022/202211/20221123/TL-WR841N(US)_V14_220816.zip)

Running On AttifyOs

dsanchez-mwb, 2023-10-12:
Same error here, any update on this issue?

megele, 2023-10-12:
This seems to be an issue with one of the programs within the firmware image. Firmadyne is oblivious to most of the things that happen inside the guest/firmware-image. Two observations seem applicable:
1.) does this message prevent the functionality in the image you try to assess? (e.g., if you're looking for web-configuration issues and the web-server runs and operates normally, you might be able to simply ignore the message).
2.) a good way to address this could be to figure out the process that's spewing the messages (e.g., via debug flags for exec) and hook up a debugger to it. Of course, you could also look at the binary statically and trace back from the error message. ymmv

cheers,
--m

dsanchez-mwb, 2023-10-12:
The problem is that it never gets to the login prompt. So according to your comment,  I can modify some system files  to disable some services/programs and then pack the firmware with these modifications and try to boot it again. Could it be a solution?

MO7HM4N, 2024-04-25:
Hey, I just found this thread after having the same issue! Did you manage to find the cause of this problem?

Saswath-A, 2025-07-22:
Same issue here , does this have any solutions or should i make a change in the firmware itself? The older version works fine but it is too old. I need to access the web interface and also i want to set a gdb server.

Saswath-A, 2025-08-24:
Ig the issue is with the version of qemu installed. So lower version can help in emulating. I would suggest you guys to use 2.5.0 qemu version. If you find any solution other than this, drop it. Hope this helps