firmadyne/firmadyne #201: Error in makeImage.sh

h4sv0c, 2022-10-22:
An error occurred when the command was executed as shown in the following figure.

[image]

This is thought to be caused by the command at line 63 of makeImage.sh.
When I executed the command on line 63, I got the following result.

[image]

However, this resulted in an error at line 64, so the error did not occur when the command was changed as shown below.

[image]

So I propose to change line 63 as follows

`TARBALL_SIZE=$(tar ztvf "${TARBALL_DIR}/${IID}.tar.gz" --totals 2>&1 |tail -1|cut -f4 -d' ')`
↓
`TARBALL_SIZE=$(tar ztvf "${TARBALL_DIR}/${IID}.tar.gz" --totals 2>&1 |tail -1|cut -f4 -d' '| sed -e 's/MiB\/s)//g')`

ddcc, 2022-10-22:
I don't think that's right, it should be grabbing the raw size and not the decompression rate. e.g. in the following:
```
Total bytes read: 1187840 (1.2MiB, 147MiB/s)
```

The goal of `tar ztvf {file} --totals 2>&1 | tail -1 | cut -f4 -d ' '` is to fetch the 1187840 field.

h4sv0c, 2022-10-22:
It seems to be due to the fact that the execution result contained Japanese characters.
[image]

Now that we know how to fix it, please CLOSE it.

Thank you for your quick response!

h4sv0c, 2022-10-22:
I'm going to CLOSE this as it has been resolved.

closed 2022-10-22 by h4sv0c