firmadyne/firmadyne #203: inferNetwork.sh didn't find any network interfaces

code2319, 2022-11-12:
I'm trying to analyze the firmware [TEW652BRPR1_FW200B0045.bin](https://github.com/code2319/Hardware-Hacking/blob/main/TEW-652BRPv2/TEW652BRPR1_FW200B0045.bin) of my old trend net router, and `inferNetwork.sh` didn't find any network interfaces, can you help me?.

workaccount23, 2022-12-01:
I'm having the same issue, is there some documentation on how InferNetwork.sh works? I'd like to spend some time to debug it but I can't find much about it

ddcc, 2022-12-02:
It should be discussed in the paper

ThomEsgate, 2023-02-15:
Any update on this?

nobodynate, 2023-04-19:
From the [paper](https://github.com/firmadyne/firmadyne/blob/master/paper/paper.pdf)
> As an inconvenient truth, improving the emulation success
rates or fixing network configuration detection for firmware
images from, e.g., Tomato by Shibby, is a manual process. It
requires an analyst to manually examine system logs in order
to identify and classify emulation failures based on root cause,
then make the changes that are necessary to support these
images. Oftentimes, this may be a cyclic process, as there can
be multiple causes of emulation failure.

I'm having the same issue with some Linksys firmware. I'm a complete newb to firmware emulation but if I figure anything out I'll share it.